ketilmo/balena-ads-b #13: [Feature Request] Implement dump978-fa UAT Support

ketilmo, 2021-01-03:


ketilmo, 2021-03-22:
Hi everyone,

This item is in progress. It's not possible to test UAT here, so I'm collaborating with a couple of persons in the US for this. I will update this ticket as soon as I have news.

Kind regards,
Ketil

jason0x21, 2021-10-12:
Are you still getting enough US help with this?  If not let me know.  My existing rig (that I'm planning on converting to balena-ads-b), has a working dual-receiver setup.

JPGMC, 2022-01-23:
Hello, if you can share the instructions, it would be great. I am upgrading to the latest version and trying to figure out how to configure UAT since we don't have access to systemctl in the container. Any tips appreciate.

ketilmo, 2022-01-24:
Hi @JPGMC,

Certainly! It's been tested by several persons now and will be released here soon. But if you want to try it straight away, feel free to contact me directly here: https://ketil.mo.land/contact

ketilmo, 2022-12-29:
This feature has now been implemented. 🥳 Please refer to the documentation for details: https://github.com/ketilmo/balena-ads-b#part-9--configure-uat-optional-and-us-only

closed 2022-12-29 by ketilmo