ketilmo/balena-ads-b #14: [Feature Request] ADS-B Exchange Integration

ketilmo, 2021-01-05:
Implement ADS-B Exchange Support, along the same lines used in the PlanerFinder integration that's part of this repo.

mirkster, 2021-01-18:
I would be happy to see this integration.

ketilmo, 2021-01-18:
Hi @mirkster, thanks for your feedback! Consider joining our mailing list to be notified about news regarding this integration. You'll find it here: https://github.com/ketilmo/balena-ads-b

ketilmo, 2021-01-18:
Hi @mirkster, thanks for your feedback! Consider joining our mailing list to be notified about news regarding this integration. You'll find it here: https://buttondown.email/balena-ads-b

marcelstoer, 2021-01-26:
Maybe the non-Balena images I built for ADS-B Exchange contain bits'n pieces you might find helpful: https://github.com/marcelstoer/docker-adsbexchange

mirkster, 2021-01-26:
I used the Belana image for under a week, it's great and I learnt a lot from it as someone starting out in ADS-B. Then decided that I really wanted to feed ADSB Exchange and that I wanted something that was not based on Belana as I like to have more control of what I run. So I deployed this image [https://www.adsbexchange.com/how-to-feed/adsbx-custom-pi-image/](url) and then just installed the different feeder clients that I actually wanted. Been using for a couple of days and it seems to be working well so far.

Now if I could just do something about getting a better place to put the antenna. Unfortunately, the best I can do is a third floor window in a five floor building with no possibility to get access to the roof.

marcelstoer, 2021-01-26:
@mirkster I understand your situation. My outdoor RPi TTN gateway has a wonderful spot on the roof of my house (powered over PoE). I thought it would be a shame if I couldn't use it as an ADS-B feeder. However, since I wanted to keep it's LoRaWAN features, also provided by a Docker image (my own), I wasn't ready to wipe the Pi just to install the ADS-B Exchange Pi image. That's the motivation behind offering those ADS-B Exchange Docker images. Some of the folks at that community weren't very pleased when I announced their availability since they find Docker complete rubbish. You wouldn't believe what sort of emails I was getting.

ketilmo, 2021-01-27:
> Maybe the non-Balena images I built for ADS-B Exchange contain bits'n pieces you might find helpful: https://github.com/marcelstoer/docker-adsbexchange

Thanks @marcelstoer, I will definitely have a look!

> . I thought it would be a shame if I couldn't use it as an ADS-B feeder. However, since I wanted to keep it's LoRaWAN features, also provided by a Docker image (my own), I wasn't ready to wipe the [...]

That's some coincidence! It was my initial motivation for creating these images, too. In fact, I have the original combined LoRaWAN-gateway and ADS-B-feeder in its own repo here: https://github.com/ketilmo/balena-ttngw-adsb

The motivation for starting using balena was the ability for easy remote administration, their write-protected file system (I experienced frequent SD-card defects when running self-installed), and easy deployment and roll-back of changes. A few years into this journey, I'm delighted and haven't look back. Besides, the balena crew is amiable and supportive, which I find to be a big bonus.

garethhowell, 2021-03-15:
I've had a go at doing this. See (https://github.com/ketilmo/balena-ads-b/pull/20)

ketilmo, 2021-03-22:
Hi @mirkster,

I'm happy to report that we now have support for ADSB Exchange thanks to the following pull request from @garethhowell (inspired by code from @marcelstoer): #20

The changes were just merged in. Please refer to the readme for details about how to configure this.

A big thanks to the guys for providing the ADSB Exchange support. 

Enjoy the feeding of yet another network! :) 

Kind regards,
Ketil

closed 2021-03-22 by ketilmo