ketilmo/balena-ads-b #15: Bad Gateway

Samuel20354, 2021-01-20:
So i have set up the abs flight tracker on balena cloud everything is setup up but when i try to access the web page it says Bad Gateway thats all 

Thanks

ketilmo, 2021-01-21:
Hi @Samuel20354, thanks for reaching out!

What webpage are you trying to access? Is it the local device, as outlined in [Step 8](https://github.com/ketilmo/balena-ads-b/blob/master/README.md#part-8--exploring-flight-traffic-locally-on-your-device) in the documentation?

Also, what device are you running on?

Samuel20354, 2021-01-21:
Hi thanks @ketilmo but I no longer need help I have now made my rpi (raspberry pi) run pi-hole

Thanks

ketilmo, 2021-01-21:
@Samuel20354 That's noted! Let us know if you give it another try in the future. :)

closed 2021-01-21 by ketilmo