ketilmo/balena-ads-b #17: Error during build process - client.planefinder.net Not Found (404)

MiBatu, 2021-02-27:
Hello,
At the stage of push to the device (_git push balena master_) there is an error with client.planefinder.net address:

> [planefinder]      Processing triggers for libc-bin (2.28-10) ...
> [planefinder]      --2021-02-27 08:52:54--  http://client.planefinder.net/pfclient_4.2.70_armhf.deb
> [planefinder]      Resolving client.planefinder.net (client.planefinder.net)... 
> [planefinder]      23.63.240.192, 23.63.240.186, 2600:141b:7000::173f:f0ba, ...
> [planefinder]      Connecting to client.planefinder.net (client.planefinder.net)|23.63.240.192|:80... 
> [planefinder]      connected.
> [planefinder]      
> [planefinder]      HTTP request sent, awaiting response... 
> [planefinder]      404 Not Found
> [planefinder]      
> [planefinder]      2021-02-27 08:52:54 ERROR 404: Not Found.
> [planefinder]      
> [planefinder]      Removing intermediate container d3d3b3cb1600
> [planefinder]      The command '/bin/sh -c chmod +x /tmp/planefinder_installer.sh && 	./planefinder_installer.sh && 	chmod +x /start.sh && 	rm -rf /tmp/*' returned a non-zero code: 8
> [Info]             Uploading images
> [Success]          Successfully uploaded images
> [Error]            Some services failed to build:
> [Error]              Service: planefinder
[Error]                Error: The command '/bin/sh -c chmod +x /tmp/planefinder_installer.sh && 	./planefinder_installer.sh && 	chmod +x /start.sh && 	rm -rf /tmp/*' returned a non-zero code: 8
> [Info]             Built on arm05
> [Error]            Not deploying release.
> 
> **remote: error: hook declined to update refs/heads/master**
> **To git.balena-cloud.com:g_michael_b2/airradar.git**
> **! [remote rejected] master -> master (hook declined)**
> **error: failed to push some refs to 'g_michael_b2@git.balena-cloud.com:g_michael_b2/airradar.git'**
> root@orangepione:~/balena-ads-b#

Can you kindly check how to fix this?
Thank you!

ketilmo, 2021-02-27:
Hi @MiBatu,

Thanks for reporting this and for sharing your error logs. The reason for this problem, it turns out, is that PlaneFinder has released a new feeder version and deleted the old one. I have now updated the balena-ads-b repo to reflect this change, and it should build OK again. Sorry for the inconvenience. Please let us know if you experience any further issues.
To stay up to date on future changes, please consider joining our mailing list: https://buttondown.email/balena-ads-b

Have a nice day,
Ketil

closed 2021-02-27 by ketilmo

MiBatu, 2021-02-27:
Hey, Ketil,
Thank you so much for your prompt response! Now it work=)

			    \
			     \
			      \\
			       \\
			        >\/7
			    _.-(6'  \
			   (=___._/` \
			        )  \ |
			       /   / |
			      /    > /
			     j    < _\
			 _.-' :      ``.
			 \ r=._\        `.
			<`\\_  \         .`-.
			 \ r-7  `-. ._  ' .  `\
			  \`,      `-.`7  7)   )
			   \/         \|  \'  / `-._
			              ||    .'
			               \\  (
			                >\  >
			            ,.-' >.'
			           <.'_.''
			             <'

Have a nice day,
Michael.

ketilmo, 2021-02-27:
That’s great to hear, @MiBatu! Enjoy your feeder and reach out if you need help again!

Best regards,
Ketil