ketilmo/balena-ads-b #18: dump1090 web-page returns "Bad Gateway"

MiBatu, 2021-03-03:
Hi, Ketil,
I hope you are doing well!

I continue to set up the my receiver and had started from FlightRadar24 (where I have the account already).
I've set FR24_KEY and checked logs to find how it is going. Unfortunately it looks like fr24feed can't connect dump1090:

_03.03.21 19:13:12 (+0300)  fr24feed  2021-03-03 16:13:12 | [feed][n]ping 25
03.03.21 19:13:13 (+0300)  fr24feed  2021-03-03 16:13:13 | [feed][n]syncing stream result: 1
03.03.21 19:13:14 (+0300)  fr24feed  2021-03-03 16:13:14 | [reader][i]Connecting to unknown receiver via (tcp://dump1090-fa:30005)
03.03.21 19:13:14 (+0300)  fr24feed  2021-03-03 16:13:14 | [reader][e]Could not connect to tcp://dump1090-fa:30005
03.03.21 19:13:19 (+0300)  fr24feed  2021-03-03 16:13:19 | [reader][i]Connecting to unknown receiver via (tcp://dump1090-fa:30005)
03.03.21 19:13:19 (+0300)  fr24feed  2021-03-03 16:13:19 | [reader][e]Could not connect to tcp://dump1090-fa:30005_

I also found that local web-page of the device returns "Bad Gateway".
And I also see strange "process killing" activity in the dump1090 log:

_03.03.21 19:07:00 (+0300)  dump1090-fa  Verifying settings...
03.03.21 19:07:00 (+0300)  dump1090-fa   
03.03.21 19:07:02 (+0300)  dump1090-fa  Receiver latitude is set: 43.5209
03.03.21 19:07:02 (+0300)  dump1090-fa  Receiver longitude is set: 39.8595
03.03.21 19:07:02 (+0300)  dump1090-fa   
03.03.21 19:07:02 (+0300)  dump1090-fa  Settings verified, proceeding with startup.
03.03.21 19:07:02 (+0300)  dump1090-fa   
03.03.21 19:07:02 (+0300)  dump1090-fa  Wed Mar  3 16:07:02 2021 UTC  dump1090-fa 4.0 starting up.
03.03.21 19:07:03 (+0300)  dump1090-fa  rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
03.03.21 19:07:03 (+0300)  dump1090-fa  Found Rafael Micro R820T tuner
03.03.21 19:07:03 (+0300)  dump1090-fa  rtlsdr: enabling tuner AGC
03.03.21 19:07:03 (+0300)  dump1090-fa  Allocating 4 zero-copy buffers
03.03.21 19:07:03 (+0300)  dump1090-fa  /start.sh: line 46:    13 Killed                  /usr/bin/dump1090-fa --device-type rtlsdr --device "$DUMP1090_DEVICE" --lat "$LAT" --lon "$LON" --fix --gain "$DUMP1090_GAIN" --ppm "$DUMP1090_PPM" --max-range "$DUMP1090_MAX_RANGE" --net --net-heartbeat 60 --net-ro-size 1000 --net-ro-interval 1 --net-http-port 0 --net-ri-port 0 --net-ro-port 30002,30102 --net-sbs-port 30003 --net-bi-port 30004,30104 --net-bo-port 30005,30105 --raw --json-location-accuracy 2 --write-json /run/dump1090-fa --quiet_

Looking forward for your advise how can I fix this.

B.R.
Michael

ketilmo, 2021-03-03:
Hi @MiBatu,

All good here, thanks! I hope you are doing well on your end.

Thanks for sharing the detailed log. What hardware are you running? Raspberry Pi 3 with 64 bit OS/image, by any chance? If that's the case, I would recommend creating a new balena image for Raspberry Pi 3 32 bit. If you're using the "Deploy with balena"-button, that image is just called "Raspberry Pi 3."

The 64-bit RPi 3 image is currently running a slightly older OS version, and there is a kernel issue here that's not been fixed yet. Hopefully, it will work in the future.

If you are not running on RPi 3 64 bit, we would need to do some more digging.

Best regards,
Ketil

MiBatu, 2021-03-05:
Hi Ketil,
You were right, the cause of this issue was in the OS/image. I've reinstalled new balena image for 32 bit and it helps.
Thank you for your help!

B.R.
Michael

closed 2021-03-05 by MiBatu