ketilmo/balena-ads-b #21: Flightaware MLAT not working

Placemaster, 2021-03-19:
The system is running and all are sending data. Only for the Flightaware (Piaware) it does not sent MLAT data. For the other services this is running normally. I have restarted the Flightaware service and than it switches on but shortly after is is red again.
Please have a look what is going on.

ketilmo, 2021-03-21:
Hi @Placemaster,

Thanks for reaching out. Sorry to hear that you are experiencing problems. There can be several different reasons why FlightAware is not accepting your MLAT data. To troubleshoot, please head over to your FlightAware ADS-B Feeder Page.

Do you see the following warning on your page?

[image]

If you do, please hold your mouse pointer over "View Anomalies" to see what FlightAware reports. Common problems include wrong coordinates (or height), high CPU usage, or high temperature.

Kind regards,
Ketil

Placemaster, 2021-03-21:
Hi Ketil,

thanks for your reply. This is the warning I have.

This is the performance of the Raspberry Pi. So everything in green, looks ok.

But my settings are ok, see below.

I do not know what to do.
Do you have some more advice (maybe on how to determine the exact location and height).

Regards,
Erwin

ketilmo, 2021-03-21:
Hi Erwin,

It looks like the images have disappeared during upload. Can you please do another attempt at adding them?

Kind regards,
Ketil

Placemaster, 2021-03-21:
Hi,

hope it works this time.

ketilmo, 2021-03-21:
Hi Erwin,

It's very strange, but I can't see your images. Can you see them yourself? Are you dragging and dropping them into the post? What happens when you click the "Preview" tab?

Kind regards,
Ketil

Placemaster, 2021-03-21:
Hi,

I did add them normally.
Please check the link, I’ll share them now.
https://1drv.ms/u/s!AjMWmIHT26pchL4tIJJsFDJaAIEzFA?e=aONDvM <https://1drv.ms/u/s!AjMWmIHT26pchL4tIJJsFDJaAIEzFA?e=aONDvM>

Hope it works.

Regards,
Erwin

ketilmo, 2021-03-21:
Hi Erwin,

When I open the link, OneDrive tells me the files are no longer available. Can you verify that the sharing settings allow for everybody to see them?

Kind regards,
Ketil

Placemaster, 2021-03-21:
Hi Ketil,

everyone with link should have acces. I use this all the time and had no complaints soar.
Let try it once more.
https://1drv.ms/u/s!AjMWmIHT26pchL4tIJJsFDJaAIEzFA?e=aONDvM <https://1drv.ms/u/s!AjMWmIHT26pchL4tIJJsFDJaAIEzFA?e=aONDvM>

If this is not working I will try to find another online location.

Regards,
Erwin

ketilmo, 2021-03-21:
Hi Erwin,

I was able to open the images now – thanks! This is useful. To further try to nail down the problem:

– What hardware are you running on?
– Is the USB dongle connected directly to the device, or through a hub?
– Can you have a look into the logs at FlightAware's webpage and see if you find any specific error messages?

Kind regards,
Ketil

Placemaster, 2021-03-21:
Hi,

I am running it on a raspberry pi 3.
The dongle is connected directly to the raspberry pi.

When I hoover over the MLAT red box it says :  MLAT disabled/not synchronized/Latitude Longitude not set

It is set by the setting I did in the balena device variables Lat and Lon.

Below the extract of the log file.

[2021-03-21 18:32 CET] mlat-client(67): Receiver: 193.0 msg/s received 38.8 msg/s processed (20%)
[2021-03-21 18:32 CET] mlat-client(67): Server status: clock unstable
[2021-03-21 18:32 CET] mlat-client(67): Server: 0.0 kB/s from server 0.0kB/s TCP to server 0.4kB/s UDP to server
[2021-03-21 18:32 CET] mlat-client(67): Receiver status: connected
[2021-03-21 18:36 CET] 1399 msgs recv'd from the ADS-B data program at dump1090-fa/30005 (383 in last 5m); 1388 msgs sent to FlightAware
[2021-03-21 18:46 CET] 2091 msgs recv'd from the ADS-B data program at dump1090-fa/30005 (316 in last 5m); 2080 msgs sent to FlightAware
[2021-03-21 18:47 CET] mlat-client(67): Receiver: 214.7 msg/s received 34.4 msg/s processed (16%)
[2021-03-21 18:47 CET] mlat-client(67): Server: 0.0 kB/s from server 0.0kB/s TCP to server 0.4kB/s UDP to server
[2021-03-21 18:47 CET] mlat-client(67): Aircraft: 6 of 7 Mode S, 11 of 11 ADS-B used
[2021-03-21 18:47 CET] mlat-client(67): Receiver status: connected
[2021-03-21 18:51 CET] 2414 msgs recv'd from the ADS-B data program at dump1090-fa/30005 (323 in last 5m); 2403 msgs sent to FlightAware
[2021-03-21 18:56 CET] 2875 msgs recv'd from the ADS-B data program at dump1090-fa/30005 (461 in last 5m); 2864 msgs sent to FlightAware
[2021-03-21 19:02 CET] mlat-client(67): Server status: clock unstable
[2021-03-21 19:02 CET] mlat-client(67): Receiver status: connected
[2021-03-21 19:02 CET] mlat-client(67): Aircraft: 5 of 5 Mode S, 17 of 18 ADS-B used
[2021-03-21 19:04 CET] mlat-client(100): Listening for Extended Basestation-format results connection on port 30106
[2021-03-21 19:04 CET] mlat-client(100): fa-mlat-client 0.2.11 starting up
[2021-03-21 19:04 CET] my feeder ID is 63905f17-762b-4576-9f2a-22d36022ee35
[2021-03-21 19:04 CET] mlat-client(100): Route MTU changed to 1500
[2021-03-21 19:04 CET] mlat-client(100): Listening for Beast-format results connection on port 30105
[2021-03-21 19:04 CET] mlat-client(100): Detected BEAST format input
[2021-03-21 19:04 CET] mlat-client(100): Beast-format results connection with 172.17.0.4:30104: connection established
[2021-03-21 19:04 CET] my feeder ID is 63905f17-762b-4576-9f2a-22d36022ee35
[2021-03-21 19:04 CET] multilateration data requested
[2021-03-21 19:04 CET] attempting to restart dump1090.. 
[2021-03-21 19:04 CET] can't restart skyaware978, no services that look like skyaware978 found
[2021-03-21 19:04 CET] UAT support disabled by local configuration setting: uat-receiver-type
[2021-03-21 19:04 CET] Started faup1090 (pid 52) to connect to the ADS-B data program at dump1090-fa/30005
[2021-03-21 19:04 CET] mlat-client(51): Listening for Extended Basestation-format results connection on port 30106
[2021-03-21 19:04 CET] Starting faup1090: /usr/lib/piaware/helpers/faup1090 --net-bo-ipaddr dump1090-fa --net-bo-port 30005 --stdout --lat 51.879 --lon 5.835
[2021-03-21 19:04 CET] mlat-client(51): Beast-format results connection with 172.17.0.4:30104: connection established
[2021-03-21 19:05 CET] 99 msgs recv'd from the ADS-B data program at dump1090-fa/30005; 99 msgs sent to FlightAware
[2021-03-21 19:10 CET] 718 msgs recv'd from the ADS-B data program at dump1090-fa/30005 (619 in last 5m); 718 msgs sent to FlightAware
[2021-03-21 19:15 CET] 1261 msgs recv'd from the ADS-B data program at dump1090-fa/30005 (543 in last 5m); 1261 msgs sent to FlightAware
[2021-03-21 19:19 CET] mlat-client(51): Server: 0.0 kB/s from server 0.0kB/s TCP to server 0.7kB/s UDP to server
[2021-03-21 19:19 CET] mlat-client(51): Receiver status: connected
[2021-03-21 19:19 CET] mlat-client(51): Server status: clock unstable
[2021-03-21 19:19 CET] mlat-client(51): Aircraft: 2 of 4 Mode S, 12 of 12 ADS-B used
[2021-03-21 19:19 CET] mlat-client(51): Receiver: 328.2 msg/s received 61.2 msg/s processed (19%)
[2021-03-21 19:20 CET] 1681 msgs recv'd from the ADS-B data program at dump1090-fa/30005 (420 in last 5m); 1681 msgs sent to FlightAware
[2021-03-21 19:25 CET] 2104 msgs recv'd from the ADS-B data program at dump1090-fa/30005 (423 in last 5m); 2104 msgs sent to FlightAware
[2021-03-21 19:30 CET] 2377 msgs recv'd from the ADS-B data program at dump1090-fa/30005 (273 in last 5m); 2377 msgs sent to FlightAware
[2021-03-21 19:34 CET] mlat-client(51): Receiver status: connected
[2021-03-21 19:34 CET] mlat-client(51): Server: 0.0 kB/s from server 0.0kB/s TCP to server 0.3kB/s UDP to server
[2021-03-21 19:34 CET] mlat-client(51): Aircraft: 1 of 3 Mode S, 15 of 15 ADS-B used
[2021-03-21 19:35 CET] 2656 msgs recv'd from the ADS-B data program at dump1090-fa/30005 (279 in last 5m); 2656 msgs sent to FlightAware
[2021-03-21 19:40 CET] 3014 msgs recv'd from the ADS-B data program at dump1090-fa/30005 (358 in last 5m); 3014 msgs sent to FlightAware
[2021-03-21 19:49 CET] mlat-client(51): Server status: clock unstable
[2021-03-21 19:49 CET] mlat-client(51): Receiver status: connected
[2021-03-21 19:49 CET] mlat-client(51): Aircraft: 8 of 9 Mode S, 14 of 14 ADS-B used
[2021-03-21 19:49 CET] mlat-client(51): Receiver: 264.4 msg/s received 42.6 msg/s processed (16%)
[... cut ...]
[2021-03-21 21:20 CET] mlat-client(51): Receiver: 309.0 msg/s received 42.2 msg/s processed (14%)
[2021-03-21 21:20 CET] mlat-client(51): Server: 0.0 kB/s from server 0.0kB/s TCP to server 0.6kB/s UDP to server
[2021-03-21 21:20 CET] 9801 msgs recv'd from the ADS-B data program at dump1090-fa/30005 (489 in last 5m); 9801 msgs sent to FlightAware

If I look at the website of Flightaware with sat and than nearby sites it says I do have MLAT (it is a green checkmark).
I am confused.

Regards,
Erwin

ketilmo, 2021-03-21:
Thanks Erwin,

This is useful. Next step is to see if you are losing packets from the USB dongle. Please follow the following steps.

1. From the balena dashboard, head back to your device's Summary page.
2. Add a new variable named DISABLED_SERVICES with the value dump1090-fa
3. Head back to your device's Summary page. Inside the Terminal section, click Select a target, then dump1090-fa, and finally Start terminal session. This will open a terminal which lets you interact directly with your PiAware container.
4. Once the terminal prompt appears, enter `rtl_test -s 2400000`, then press return.
5. Let the script run for a while, copy the results and paste them as a post here.

To reset to normal operations:
1. Head back to your device's Summary page.
2. Delete the variable named DISABLED_SERVICES.
3. Wait for the containers to restart.

Kind regards,
Ketil

Placemaster, 2021-03-21:
Hi,

this is the output:

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2400000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 88 bytes
lost at least 20 bytes
lost at least 20 bytes
lost at least 164 bytes

So it is loosing bytes. How to prevent that?

Regrads,
Erwin

ketilmo, 2021-03-21:
Hi Erwin,

Is it losing bytes consistently, or only during startup? 

See this forum thread for more information: https://discussions.flightaware.com/t/mlat-issues-clock-unstable/42083/3
There are several posts around the same symptoms. Google "Server status: clock unstable" to find more.

You could try to switch the dongle over to a different USB port and see if that helps. Also, an unstable PSU can cause issues.

I noticed that you latitude value only had four decimal points. Increasing to five could be worth trying, with the help of https://maps.google.com. (This is not related to the sample drop, but could affect the clock syncing during multilateration.)

Kind regards,
Ketil

Placemaster, 2021-03-22:
Hi Ketil,

I have changed the power supply for a 3 Amp version. Now i am loosing at startup 64 bytes and  none after.
It looks like it has solved my problem. Thanks for your help.

Regards,
Erwin

ketilmo, 2021-03-22:
Hi Erwin,

That's great news! I am happy to hear that you were able to solve the issue with a 3 amp PSU. Hopefully, your device will run stable from now on. Happy feeding!

Kind regards,
Ketil

closed 2021-03-22 by ketilmo