ketilmo/balena-ads-b #25: Can´t get data from port 30005 or 30106 for VRS (Virtual Radar Server)

DanBuBerlin, 2021-03-21:
Hi,

Thank you for your great work. I wanted to set up a new Pi4 as an ADS-B receiver. So far, it has always been very laborious and time-consuming. With your Balena package, this is almost child's play. TOP!

I normally pull the data for my VRS (Virtual Radar Server) from my dump1090 receivers via ports 30005 and 30106, but with my actually Balena installation I don't seem to have any access?
Are the ports not accessible from the outside? Is there any other way to push to a VRS? 

Thanks

Daniel

ketilmo, 2021-03-22:
Hi @DanBuBerlin,

Thanks for reaching out and for your kind feedback. It's motivating to hear that you are enjoying your balena ADS-B setup.

Your feature request is appreciated. And luckily, exposing the exposed ports is simple. I'm happy to report that I just pushed a change that hopefully will give you what you need (through 546b9cc797e9f90c355dcddd78755b3bd509ba98). Try to update your installation to the newest release. You should be able to connect VRS to your local balena ADS-B device.

Let us know how it goes!

Kind regard,
Ketil

closed 2021-03-22 by ketilmo

DanBuBerlin, 2021-03-22:
Hi @ketilmo ,

Thanks for your great work!
I have now also watched the video about the project and see that we both have similar interests and goals. Great. Mine are the helicopters. I am currently managing several receivers remotely.
As a newcomer to Balena, have I not yet found the point where I can update the system or with what click? Can you give me a hand there?

Is there a basic manual where I can look up these things? To be honest, I feel a bit stupid :-)

Greetings from Berlin

Daniel