ketilmo/balena-ads-b #26: [Feature Request] Ability to push data to VRS (Virtual Radar Server)

DanBuBerlin, 2021-03-21:
It would be nice if a way to push data to VRS (Virtual Radar Server) could be created. This means that the receiver could be installed in a remote location.

DanBuBerlin, 2021-03-22:
A friend told me that installing ModeSMixer2 could be a way?

DanBuBerlin, 2021-03-23:
ModeSMixer2 on GitHub
https://github.com/abcd567a/mm2/blob/master/README.md

An interesting thread:
https://forum.flightradar24.com/forum/radar-forums/flightradar24-feeding-data-to-flightradar24/11406-how-to-install-modesmixer2-on-rpi-and-get-vrs-like-range-plot

Here are more informations:
https://sonicgoose.com/combining-and-rebroadcasting-ads-b-feeds-with-modesmixer2/

sebadv, 2021-04-16:
I don't really get it. I'm using Virtual Radar Server. Just entered the IP of the pi and it works on port 30002.