ketilmo/balena-ads-b #27: repos/balena-io/wifi-connect has moved, causing build error.

drewgolden, 2021-04-05:
curl https://api.github.com/repos/balena-io/wifi-connect/releases/latest -s 
{
  "message": "Moved Permanently",
  "url": "https://api.github.com/repositories/37727198/releases/latest",
  "documentation_url": "https://docs.github.com/v3/#http-redirects"
}

closed 2021-04-05 by drewgolden

ketilmo, 2021-04-05:
Hi @drewgolden,

Thanks for engaging in balena ADS-B. Were you able to build the repository without any error messages?

Kind regards,
Ketil

drewgolden, 2021-04-05:
I just branched and removed wifi-connect from the docker compose to get the
build to work for me.

---
Drew Golden

ketilmo, 2021-04-08:
With help from Anders in the community, I was able to reproduce and solve this issue. It turns out the wifi-connect repo's URL was changed. Now everything should work fine again. Thanks to @drewgolden for reporting this and to Anders for supplying the fix.

commit a73863d0 (2021-04-08): Fixed #27.