ketilmo/balena-ads-b #33: Adding ADSBexchange’s dashboard

JPGMC, 2021-07-18:
It would be great if you could integrate ADSB Exchange's Graphana and display the unfiltered local feed.

ketilmo, 2021-07-22:
Hi @JPGMC,

I just updated the ADSB Exchange integration to include their stats, among other things. You should also be able to view your data feed through their tar1090 viewer. After updating to the latest release, you'll find everything on ADSB Exchange's [Feeder Status](https://www.adsbexchange.com/myip/) page.

Let me know how this works for you.

Kind regards,
Ketil

JPGMC, 2021-07-22:
Hello @ketilmo ,

Worked for me, thank you. If I may add, a new endpoint on the page would be a great addition (Public Device URL, same as FR24, skyward and plane finder.

Cheers and thanks for the quick turn around!