ketilmo/balena-ads-b #34: After Balena OS update (balenaOS 2.82.10+rev1):  dump1090-fa  rtlsdr: no supported devices found.

sebadv, 2021-07-30:
dump1090-fa  Receiver longitude is set: 3.7641157410193986
 dump1090-fa   
 dump1090-fa  Settings verified, proceeding with startup.
 dump1090-fa   
 dump1090-fa  Fri Jul 30 18:58:26 2021 UTC  dump1090-fa 5.0 starting up.
 dump1090-fa  rtlsdr: no supported devices found.
 dump1090-fa  Verifying settings...
 dump1090-fa   
 dump1090-fa  Receiver latitude is set: 51.04xxxxxxxxxxxxxxxx
 dump1090-fa  Receiver longitude is set: 3.76xxxxx
 dump1090-fa   
 dump1090-fa  Settings verified, proceeding with startup.
 dump1090-fa   
 dump1090-fa  Fri Jul 30 18:59:31 2021 UTC  dump1090-fa 5.0 starting up.
 dump1090-fa  rtlsdr: no supported devices found.

I'm using the flightaware pro stick plus (since april 2021) on a Raspberry Pi4 4GB

ketilmo, 2021-08-03:
Hi @sebadv,

I'm sorry to hear that you are having issues. I have just updated several of my testbed units (both Raspberry Pi 3 and 4s) to the latest balenaOS version, and they all work as expected. Have you tried to do a full power off of the system or tried swapping the RTLSDR stick to a different USB port?

Kind regards,
Ketil

sebadv, 2021-08-03:
Hi @ketilmo  
Yes, did several power cycles. Swapping USB ports, changing between FlightAware stick and my previous SDR stick.
Still the same error.
The only thing left is to do a completely new install, I'm going to do that this evening.

ketilmo, 2021-08-03:
Hi @sebadv,

Before reinstalling, can you try this :
1. From the balena dashboard, head back to your device's Summary page.
2. Add a new variable named `DISABLED_SERVICES` with the value `dump1090-fa`
3. Head back to your device's Summary page. Inside the Terminal section, click Select a target, then dump1090-fa, and finally Start terminal session. This will open a terminal that lets you interact directly with your PiAware container.
4. Once the terminal prompt appears, enter `rtl_test -s 2400000`, then press return.

Does the device respond to this command?

Kind regards,
Ketil

sebadv, 2021-08-03:
Hi @ketilmo 
I tried with both usb sticks on several usb-ports. Everytime I get: No supported devices found.
I suspect there is something corrupt.

Oh well, a fresh install isn't that bad :-) I'll keep you updated

ketilmo, 2021-08-03:
Hi @sebadv,

Thanks! Fingers crossed you will be able to fix it. When reinstalling, you have the option to downgrade to an older balenaOS version.

Best regards,
Ketil

sebadv, 2021-08-03:
I'm up and running again on the previous BalenaOS version. I'll try to upgrade again to see if the problem will reoccur.

ketilmo, 2021-08-03:
Thanks for the update, @sebadv! That's great to hear. But also very strange. I guess that there might be subtle hardware differences between our boards. If you can reproduce the issue, it's probably a balenaOS bug that should be reported to the core team. Let me know how it goes!

Ketil

sebadv, 2021-08-03:
After the upgrade I again receive the following message from dump1090-fa:

dump1090-fa  Settings verified, proceeding with startup.
 dump1090-fa   
 dump1090-fa  Tue Aug  3 14:23:10 2021 UTC  dump1090-fa 5.0 starting up.
 dump1090-fa  rtlsdr: no supported devices found.

ketilmo, 2021-08-03:
Thanks for the update, @sebadv! Would you mind doing a clean install straight to the latest version, without upgrading?

Kind regards,
Ketil

sebadv, 2021-08-03:
The same error:

 dump1090-fa  Tue Aug  3 14:48:42 2021 UTC  dump1090-fa 5.0 starting up.
 dump1090-fa  rtlsdr: no supported devices found.

ketilmo, 2021-08-03:
Thanks @sebadv! That's a bummer, but it's great that you've been able to pinpoint the issue. Is it correctly understood that this issue was introduced by balenaOS version 2.82.10+rev1. And it worked in the version prior to that (balenaOS 2.80.xx)? Can you send me the full specs of your board? Then, I can report this problem to the balena core team. Alternatively, you are welcome to do it directly through the Product Support thread at the balena forums: https://forums.balena.io/c/product-support/42

Thanks again for debugging this, @sebadv!

Kind regards,
Ketil

sebadv, 2021-08-03:
Hi @ketilmo 

it's a Raspberry Pi 4 - 8GB - Model B (BCM2711)
Do you need anything else? If so, please tell me how to get the info. I'm not a huge hardware wizard ;-)

ketilmo, 2021-08-05:
Thanks @sebadv, that should be sufficient. I have now reported the issue in the balena forums: https://forums.balena.io/t/rpi-4-issues-after-update-to-balenaos-2-82-10-rev1/342914. Stay tuned for updates.

ketilmo, 2021-10-02:
balena's core team has now fixed the problem (https://github.com/balena-os/balena-raspberrypi/issues/695), so I'm marking this issue as resolved.

closed 2021-10-02 by ketilmo