ketilmo/balena-ads-b #39: Question in regards to adsbexchange integration

wiedehopf, 2021-12-10:
https://github.com/ketilmo/balena-ads-b/blob/48ad3b5389a16e81aa6893682d5a9ecdefbb7a0c/adsb-exchange/Dockerfile.template#L64

So ... this mlat-client will clash with the normal mlat-client package as the package isn't really named different.
Python is a mess anyhow that's why the adsbexchange feed client uses a venv and just installs like that.
See this script here: https://github.com/adsbxchange/adsb-exchange/blob/master/update.sh#L180

Anyhow the normal mlat-client will work but our fork is strongly preferred.
(and our fork probably also works for other aggregators but that is in no way guaranteed)

I can look into making it adsbexchange-mlat-client i suppose including package name and everything ...

https://github.com/ketilmo/balena-ads-b/blob/48ad3b5389a16e81aa6893682d5a9ecdefbb7a0c/adsb-exchange/Dockerfile.template#L24

Are you opposed to just using the master branch?
The version used here is very very very old (it's from before the fork).
Actually this made me realize i should just remake that repository and fork from my repository ... it doesn't have all the database baggage making it huge.
But that would also mean your build will stop working.

If you want to not use a branch but rather a version ... the current commit is as good as any: 8468e346408b3c79ce16cf33e86640d03727fc4f

I'd make a PR but the changes depend on how you want to handle the above :)

ketilmo, 2021-12-11:
Hi @wiedehopf,

Thanks for reaching out, it's appreciated. Also thanks for all your contributions to the flight tracking community.

The reason why I have used fixed versions/commits is that some of the tracking packages have made breaking changes in the past. Since balena-ads-b runs on different types of hardware, I try to test releases on as many different devices as possible before upgrading. When all look good, I "shrinkwrap" the build with specific releases or commit IDs.

Breaking changes might not be an issue with ADSB Exchange, however, and I would be more than happy to follow your suggestions here. What do you consider to be the best approach? Creating a separate package with dedicated versioning, or pulling the source code straight from your master branch?

When I made the ADSB Exchange integration, I did my best to closely follow your setup as documented in the install script. But this is indeed a while ago, so please let me know if you see anything else in the code that you would like me to change.

Thanks again and kind regards,
Ketil

ketilmo, 2021-12-21:
Suggested changes have now been implemented by @wiedehopf (thanks!), closing ticket.

closed 2021-12-21 by ketilmo