ketilmo/balena-ads-b #43: Range decreased..

sadeedp, 2022-01-06:
[image]
range decreased 176 to 26nm with update

ketilmo, 2022-01-08:
Hi @sadeedp,

This issue could be related to the new auto gain feature in PiAware. You can revert to the former default setting by adding an environment variable `DUMP1090_GAIN` with the value of `-10`.

Let us know how it works for you.

jason0x21, 2022-06-17:
> I'm seeing this as well since I switched over from a regular Piaware installation on top of Raspian. I tried adding the variable to no avail. How long has the new auto gain feature been around?

This appears to have been due to a bad PoE Hat not providing enough power and failing after a week.  A new hat seems to have cured the issue.

ketilmo, 2022-06-18:
Thanks for the update, @jason0x21!

ketilmo, 2022-12-29:
I have added some more information about gain settings to the documentation: https://github.com/ketilmo/balena-ads-b#setting-dump1090-antenna-gain.

This issue will now be closed. Please reopen it if you still need help with the gain settings.

closed 2022-12-29 by ketilmo