ketilmo/balena-ads-b #51: "View live data" link on flightaware.com page takes me to wrong link

robertreid, 2022-04-25:
When I click on "Web interface: View live data" link on my [https://flightaware.com/adsb/stats/user/medicusnocte#stats-131113](https://flightaware.com/adsb/stats/user/medicusnocte#stats-131113) Flightaware status page, it takes me to wrong link: http://172.17.0.5:8080/

The device is actually on http://10.0.1.187:8080/ and when I go there, the live data comes up correctly.

How do I get my instance of balena-ads-b to tell Flightaware what the correct local link is?

ketilmo, 2022-06-18:
Hi @robertreid,

Piaware is picking ut the IP address that's used internally in the virtualized Docker setup, not the IP on the device. I'm not aware of any way to work around this currently, other than manually modifying the URL as you have done – or by using the Skyaware Anywhere link that shows all of your receivers.

ketilmo, 2022-12-29:
As there is no way to fix this, I'm closing this issue.

closed 2022-12-29 by ketilmo