ketilmo/balena-ads-b #54: Flightradar24 requiring MLAT to be disabled on custom receivers

schubydoo, 2022-05-25:
Just received an email from Flightradar24. They are asking for all receivers (that aren't directly from them) to disable MLAT if said receivers are sharing to multiple networks.

https://forum.flightradar24.com/forum/radar-forums/flightradar24-feeding-data-to-flightradar24/213550-disable-mlat-if-sharing-with-other-networks

Is there a balena variable we can set for this?

suggestbest, 2022-05-25:
You can disable this from the status page of your feeder:

https://*ip-address-of-your-feeder*:8754
- Click "Settings"
- Choose the dropdown to "disable" next to MLAT

[image: image.png]

schubydoo, 2022-05-26:
Deployed the latest and confirmed MLAT is off.

ketilmo, 2022-06-18:
Thanks for confirming, @schubydoo!

closed 2022-06-18 by ketilmo