ketilmo/balena-ads-b #70: Device always disconnects WiFi after a day or two

rblenkinsopp, 2022-12-22:
**Hardware in Use:**

- Raspberry Pi 3 Model B+
- FlightAware Pro Stick Plus USB Dongle

**Steps To Reproduce:**

- Device has Balena configuration for my Wifi as part of the device setup.
- Observe it registering correctly with all the ADS-B services.
- Leave it a day or 2

**Expected Behaviour:**

- Device continues to operate as expected and connected to the WiFi and the services.

**Actual Behaviour:**

- The device at some point disconnects from the WiFi and starts broadcasting the "balenaWiFi" network.
- The device stays in this state until power cycled, or the "wifi-connect" interfaces is used to reconnect manually.

**Comments:**

This is highly frustrating and I'd like to be able to resolve this as soon as possible!

ketilmo, 2022-12-29:
Hi @rblenkinsopp,

Thanks for reaching out and providing a detailed description of the issue you are experiencing.

The wifi-connect module will check for an Internet connection every 120 seconds. If the connection is down, the balenaWiFi hotspot will appear to let you reconfigure your WiFi settings if necessary. The balenaWiFi hotspot might be a symptom of connectivity issues with the router or upstream connectivity issues.

To debug the problem, you can try disabling the balenaWiFi service using device variables in the balena console. Create a new device variable named `DISABLED_SERVICES`, with the value `wifi-connect`. This will disable the hotspot functionality altogether.

Please let us know how this works for you.

misko85, 2023-01-03:
I have similar problems and wanted to try if it helps to turn off the WiFi service. With the device variables it does not work for me. Even after a restart WiFi remains active. I have now turned off the WiFi service manually. 

[image]

[image]

Hardware in Use:
Raspberry Pi 4
RadarBox-FlightStick

rblenkinsopp, 2023-01-03:
Thanks @ketilmo; I've done that and will report back in a few days to see if it's solved the problem.

I believe there might be a momentary connectivity drop that causes the hotspot to disconnect; however, should the device not reconnect when connectivity is restored without requiring manual intervention?

schubydoo, 2023-01-08:
I don't want to conflate two seperate issues, but I think wifi-connect (and maybe kiosk?) do not look for the DISABLED_SERVICES variable.  I'm looking at code to see.

ketilmo, 2023-01-08:
Thanks for pointing out this, @schubydoo. And yes, you are right. Since wifi-connect and kiosk are now implemented using balena blocks, they won't be disabled by the DISABLED_SERVICES variable. Sorry for the inconvenience, @rblenkinsopp.

And: You're right about the device reconnecting when the connection is restored. Some underlying issues might prevent this from working as it should. Do you have a spare device that you can use to test?

ketilmo, 2023-01-08:
I have revised the documentation to indicate that the wifi-connect and kiosk services cannot be disabled: c6081ff6b7101691d0ddb08b157cc1883e4171bf

schubydoo, 2023-01-08:
@rblenkinsopp I did some digging and found something that may help: https://github.com/balena-os/wifi-connect/issues/414#issuecomment-1248471330

Try adding adding 60, or 120, to the `ACTIVITY_TIMEOUT` variable in the Balena Dashboard and see if that helps.

I'm also testing some local code to see if we can pass the DISABLED_SERVICES variable to these containers.

schubydoo, 2023-01-08:
> @rblenkinsopp I did some digging and found something that may help: [balena-os/wifi-connect#414 (comment)](https://github.com/balena-os/wifi-connect/issues/414#issuecomment-1248471330)
> 
> Try adding adding 60, or 120, to the `ACTIVITY_TIMEOUT` variable in the Balena Dashboard and see if that helps.
> 
> I'm also testing some local code to see if we can pass the DISABLED_SERVICES variable to these containers.

After a bit of testing, adding the ability to disable these services would require much more coding than my current knowledge level allows.  Curious if ACTIVITY_TIMEOUT works as an official solution for the time being.  It might be possible to move away from blocks and use the core images for those two services in the future as a potential cleanup/custom changes solution.

rblenkinsopp, 2023-01-18:
Sorry for the delay in getting back, the 'DISABLED_SERVICES' method appears to have resolved this issue for me just fine, I've therefore not tested the other solution in this thread yet.

ketilmo, 2023-01-25:
Thanks for reporting back, @rblenkinsopp! Great that you got it to work. Thanks for using balena-ads-b. And thanks for the research and help, @schubydoo!

closed 2023-01-25 by ketilmo

newgas, 2023-09-24:
I came across this closed post while trying to find a solution on how to disable the wifi-connect and balenwifi as I have my adsb pi connected to ethernet.

This appears to work, applied manually via Host OS terminal and rfkill command
https://fin-docs.balena.io/1.1.x/developers/#controlling-bluetooth-and-wifi

ketilmo, 2023-09-24:
Thanks for sharing your findings, @newgas! Much appreciated!