ketilmo/balena-ads-b #75: no ADS-B data program seen listening on port 30978

g0Idfinger, 2023-02-01:
piaware keeps reporting this:
no ADS-B data program seen listening on port 30978

it never reports up any UAT traffic.  

Running pi3b with 2 SDRs configured.

ketilmo, 2023-03-02:
Hi @g0Idfinger,

Thanks for reaching out. That log message might indicate that the dump978 service is not running. Do you see any log output from the `dump978-fa` service?

ketilmo, 2023-05-18:
Hi @g0Idfinger,

Any update on this issue? Are you still having problems, or can this issue be closed?

ketilmo, 2023-09-10:
Closing due to lack of feedback. Please reach out again if you are still having issues.

serviceman77, 2024-04-14:
This is great work. I just built this on my Pi, and I am having the same issue as above. This is the piaware error. 
 piaware  attempting to start dump978.. 
 piaware  can't start dump978, no services that look like dump978 found
UAT_ENABLED	true All services are in my variables.
Radarbox is able to use the 978. Only piaware has this issue.
Thanks again

ketilmo, 2024-04-19:
@serviceman77

ketilmo, 2024-04-19:
Hi @serviceman77,

Thanks for reaching out. Can you please filter the logs on the dump978 service and see if you see any errors?

serviceman77, 2024-04-19:
[Image.png]
It was empty when I first opened it. I restarted it and now I get some logs. RadarBox is receiving but FA does not. Thanks for looking into this. Added some screenshots from FA and RB
[Image.png]
[Image.png]

serviceman77, 2024-04-20:
[image]

Radarbox
[image]

FlighyAware
[image]

ketilmo, 2024-04-28:
Thank you so much for the images, @serviceman77!

I have located the problem and pushed for a fix. Can you update your installation to the latest version and report if it resolves the problem for you?

serviceman77, 2024-04-28:
I am new to balena, how do I update?

ketilmo, 2024-04-28:
Check out the instructions here: https://github.com/ketilmo/balena-ads-b/?tab=readme-ov-file#part-14--updating-to-the-latest-version

If you deployed using the blue button, click it again, log in and then click the "use an existing fleet instead" option. Then, overwrite your current version. The settings variables will be preserved.

Don't hesitate to ask if you have more questions. Let me know how it goes.

serviceman77, 2024-04-29:
The update has solved the issue! Thanks for your excellent work.

ketilmo, 2024-04-29:
So glad to hear that, @serviceman77! Thanks for reporting back and confirming the fix. As I live in Norway and we don't have UAT rolled out here, it's quite hard to debug. So don't hesitate to reach out again in the future if you see other issues. Happy feeding!

Fixed by e4c81045961603a7b37a2e9ee8c4bdd5b1e4b540.

closed 2024-04-29 by ketilmo