ketilmo/balena-ads-b #78: [Feature Request] Add adsb.fi as a target.

jason0x21, 2023-03-09:
It looks like the ADSB Exchange breakup has splintered in a few places, and adsb.fi is one of them.  Could this be added as well as a target?  Thanks!

cws82us313, 2023-03-28:
> Short version if Ketil is intressed :) I hope he is! I´m not a coder but I do understand docker and scripts.
> 
> Use sdr-enthusiasts docker images they are in a separate class and easy to implement. No need to maintain end they have realy good support in their Discord channel.
> 
> I´m running this on Pi4 and X86_64 (intel nuc)
> 
> If interested I have config for tar1090+graphs with sdr-enthusiasts images :) basic the same method as described below with multifeeder.
> 
> Simpel howto:
> 
> git clone into local map multifeed, rootsfs folder and docker file is important.
> 
> The "run" file in /rootfs/etc/services.d/readsb)/run need the "LONG" changed to "LON" since balena-adsb use LON as Longitude parameter and sdr-enthusiasts use LONG as Longitude parameter or simply add an extra "Device Parameter" LONG same as LON uggly solution but doable.
> 
> Then in docker-compose.yml in balena root add: READSB_NET_CONNECTOR and MLAT_CONFIG are super important they are in separate lines ended with ; then choose services to feed, I added all but it simple to remove or add others.
> 
> multifeed: depends_on: - dump1090-fa image: multifeed build: ./multifeed restart: always environment: - LAT= - LON= - ALT= - READSB_NET_CONNECTOR= dump1090-fa,30005,beast_in; feed.adsb.fi,30004,beast_reduce_plus_out; feed.adsb.one,64004,beast_reduce_plus_out; in.adsb.lol,30004,beast_reduce_plus_out; feed.theairtraffic.com,30004,beast_reduce_plus_out; feed.planespotters.net,30004,beast_reduce_plus_out; data.avdelphi.com,24999,beast_out; - MLAT_CONFIG= feed.adsb.fi,31090,39000; feed.adsb.one,64006,39001; in.adsb.lol,31090,39002; feed.theairtraffic.com,31090,39003; mlat.planespotters.net,31090; expose: - "39000" - "39001" - "39002" - "39003" - "31090" tmpfs: - /run/readsb - /var/log
> 
> Then push to balena :)

Hey 
Thanks so much can you upload the code to github so it just be easy as push it ?? 
For theairtraffic thanks

kk-sonata, 2023-03-28:
> Hey Thanks so much can you upload the code to github so it just be easy as push it ?? For theairtraffic thanks

I got it working based on @drhanke's instructions (thanks!), feel free to use this branch: https://github.com/kk-sonata/balena-ads-b/tree/plus-theairtraffic

I'm a git novice but could submit a PR if I can figure it out.

cws82us313, 2023-03-29:
> I got it working based on @drhanke's instructions (thanks!), feel free to use this branch: https://github.com/kk-sonata/balena-ads-b/tree/plus-theairtraffic
>
> I'm a git novice but could submit a PR [...]

Thank you soo much it works great

cws82us313, 2023-03-29:
Just one issue seems like theairtraffic not getting my MLAT feed but other website can seems odd

drhanke, 2023-03-29:
Wow :) happy it worked

Odd with theairtraffic tho, just checked my own feeds and no problem. Guess that this page confirming that no mlat is recieved ?
https://theairtraffic.com/feed/myip/

If you select multifeeders log in balena and restart multifeeder, is the log telling any intresting info about connection to theairtraffic ?

cws82us313, 2023-03-30:
getting an error message saying  docker-multifeeder  [2023/03/29 22:56:38][readsb] ERROR: Not a valid UUID: /usr/local/share/adsbexchange/adsbx-uuid
 docker-multifeeder  [2023/03/29 22:56:38][readsb] Use this command to fix: sudo uuidgen > /usr/local/share/adsbexchange/adsbx-uuid
seems odd When i do the command sudo  uuidgen > /usr/local/share/adsbexchange/adsbx-uuid it says  No such file or directory. i done cat  /proc/sys/kernel/random/uuid and did a random number but that error message still shows up

drhanke, 2023-03-30:
Ah  UUID and MLAT_USER is missing, just add  in balena "Device Variables". Sorry missed to write about them! My bad :)
UUID use same uuid as used to ADSBX or just copy text from console "cat  /proc/sys/kernel/random/uuid"

MLAT_USER set to your radarname

Read more about them 
https://github.com/sdr-enthusiasts/docker-multifeeder

cws82us313, 2023-03-30:
mlat-client: error: argument --lat: expected one argument is the new error

drhanke, 2023-03-31:
Diffrent forks don´t use same parameters, maybe need to learn gitbug to suggest some more changes so the diffrent containers  ha the same parameters :)

In the mlat run file changes need to be done to... I run older code, but when I check now they have changed, my code has three different.

/rootfs/etc/services.d/mlat-client/run

MLAT_PARAM+=(--lat "${READSB_LAT}")
MLAT_PARAM+=(--lon "${READSB_LON}")
MLAT_PARAM+=(--alt "${READSB_ALT}")

change to: (remove READSB)
MLAT_PARAM+=(--lat "${LAT}")
MLAT_PARAM+=(--lon "${LON}")
MLAT_PARAM+=(--alt "${ALT}")

cws82us313, 2023-03-31:
Works great and we are now feeding :) thank you soooo much

ketilmo, 2023-05-18:
Great to hear that you were able to solve this, guys! It's been rather hectic at my end recently, so I have not been able to look into all the new ADSBx clones yet. But I will do it in the future when I have more time available. In the meantime, keep up the great work!

shawaj, 2023-11-21:
@ketilmo I would be happy to contribute a pr on this and the other service addition requests. But one thing I was thinking is that for any adsb-exchange clones, it maybe doesn't make sense to have very-similar containers per service?

maybe better to just have the single adsb-exchange container, and then populate different env variables which adds additional --net-connector outputs?

For compatible clones, with no additional installation stuff, it probably is not worth the extra bloat/overhead of several identical containers I guess. What do you think?

ketilmo, 2023-11-21:
@shawaj I agree entirely with you, @shawaj. It's the best approach to support the increasing number of ADSBx clones that are appearing and which are all similar in their technical nature. And yes, pull requests are appreciated. 🥳

shawaj, 2023-11-26:
I've got these all working nicely for adsb. Just adding the MLAT and then will submit a PR 👌

closed 2024-09-16 by shawaj

shawaj, 2024-09-17:
@jason0x21 @kk-sonata @cws82us313 @drhanke this is all in there now just FYI

PR #95 (2024-09-16): feat: add adsb exchange clones
Add the following ADS-B Exchange clones:
- [adsb.fi](https://adsb.fi/)
- [ADSB.lol](https://adsb.lol/)
- [ADS-B One](https://adsb.one/)
- [airplanes.live](https://airplanes.live/)
- [Planespotters.net](https://www.planespotters.net/)
- [TheAirTraffic](https://theairtraffic.com/)
- [AvDelphi](https://www.avdelphi.com/)
- [HP Radar](https://adsb.lol/)
- [RadarPlane](https://radarplane.com/)
- [Fly Italy ADSB](https://flyitalyadsb.com/)

Also:
- updates ADSB Exchange server addresses as per https://github.com/ADSBexchange/feedclient/blob/643e7ba93e445435e1037406edf9 [... cut]
commit 22f2236c (2024-09-16): feat: add adsb exchange clones

Add the following ADS-B Exchange clones:
- [adsb.fi](https://adsb.fi/)
- [ADSB.lol](https://adsb.lol/)
- [ADS-B One](https://adsb.one/)
- [airplanes.live](https://airplanes.live/)
- [Planespotters.net](https://www.planespotters.net/)
- [TheAirTraffic](https://theairtraffic.com/)
- [AvDelphi](https://www.avdelphi.com/)
- [HP Radar](https://adsb.lol/)
- [RadarPlane](https://radarplane.com/)
- [Fly Italy ADSB](https://flyitalyadsb.com/)

Also:
- adds the ability to arbitrarily define similar new services.
- updates ADSB Exchange server addresses as per https://gith [... cut]