ketilmo/balena-ads-b #88: OS update failed

Placemaster, 2023-11-02:
Since a few weeks I am not able to update my balena ADS-B device anymore.
It is a raspberry pi 3 and is running Supervisor Version15.0.0 and Host Os Version 4.0.16.
I want to update the system to SV 15.0.2 and Host OS Version 4.1.1. I select i the dashboard summarythe SV and select 15.0.2. It start to apply the update but it always fails.

After that I tried to create a new release which instaled correctly but did not update to newer versions of the SV nor the Host OS.

What can I do to get the latest update installed and working?

ketilmo, 2023-11-05:
Hi @Placemaster,

I'm sorry to hear you are having issues updating balenaOS.

To figure out why it's failing, please attempt the following:

1. Head to the **Summary** tab for the device in your balena dashboard.
2. In the **Terminal** window, click the **Select a target** dropdown, then choose **Host OS**, then click **Start terminal session.**
3. In the terminal, enter `cd` /mnt/data/balenahup/`, then `ls -la`.
4. You will see a list of log files. You can find the most recent one by looking at the timestamps.
5. View the logfile contents by issuing this command: `cat upgrade-2.x.sh.xxxxx.log (your logfile name).`

Hopefully, this will reveal some clues about what's going wrong. Feel free to share it here, redacting any sensitive or personal information.

Placemaster, 2023-11-07:
Hi,

I have managed to do the update.
In the configuration setting I have activated :Override existing update lock(s) if your app is stuck in an invalid state under an update lock <>	

After this I was able to perform the update. Now everything is up-to date again.
Thanks for reaching out.

Best regards,
Erwin

ketilmo, 2023-11-12:
Hi @Placemaster,

It's great to hear that you were able to resolve the problem. Thanks for sharing your solution, too. Happy feeding!

Kind regards,
Ketil

closed 2023-11-12 by ketilmo