ketilmo/balena-ads-b #97: Open Sky error

cws82us313, 2023-12-26:
opensky-network  Statistics
 opensky-network   - started on Mon Dec 25 10:00:33 2023
 opensky-network   - running since   1d, 00h:27m:48s
 opensky-network  
 opensky-network   Receiver
 opensky-network   -                      221289 frames received (2.51 /s, 0.93 /s)
 opensky-network   -                       88127 [ 40%] Mode-A/C (1.00 /s, 0.93 /s)
 opensky-network   -                       43185 [ 20%] Mode-S Short (0.49 /s, 0.00 /s)
 opensky-network   -                       89977 [ 41%] Mode-S Long (1.02 /s, 0.00 /s)
 opensky-network     -                       10539 [  8%] type 0 (0.12 /s, 0.00 /s)
 opensky-network     -                        3893 [  3%] type 4 (0.04 /s, 0.00 /s)
 opensky-network     -                        3312 [  2%] type 5 (0.04 /s, 0.00 /s)
 opensky-network     -                       25441 [ 19%] type 11 (0.29 /s, 0.00 /s)
 opensky-network     -                         279 [  0%] type 16 (0.00 /s, 0.00 /s)
 opensky-network     -                       72364 [ 54%] type 17 (0.82 /s, 0.00 /s)
 opensky-network     -                          18 [  0%] type 18 (0.00 /s, 0.00 /s)
 opensky-network     -                       12275 [  9%] type 20 (0.14 /s, 0.00 /s)
 opensky-network     -                        5041 [  4%] type 21 (0.06 /s, 0.00 /s)
 opensky-network   -                           0 [  0%] status (0.00 /s, 0.00 /s)
 opensky-network   -                           0 [  0%] unknown (0.00 /s, 0.00 /s)
 opensky-network   -                       88127 [ 40%] filtered (1.00 /s, 0.93 /s)
 opensky-network   -                           0 [  0%] Mode-S filtered (0.00 /s, 0.00 /s)
 opensky-network   -                      221289 unsynchronized frames
 opensky-network  
 opensky-network   Buffer
 opensky-network   -                           0 queued messages (current)
 opensky-network   -                          21 queued messages (max)
 opensky-network   -                           0 discarded messages (current)
 opensky-network   -                           0 discarded messages (overall)
 opensky-network   -                           0 discarded messages (in worst overflow situation)
 opensky-network   -                         199 messages in pool (Usage 0.00%)
 opensky-network   -                           0 dynamic pools (current)
 opensky-network   -                           0 dynamic pools (overall)
 opensky-network   -                           0 dynamic pools (max)
 opensky-network   -                           0 uncollected dynamic pools
 opensky-network   -                           6 flushes
 opensky-network   -                           0 Garbage Collector runs
 opensky-network  
 opensky-network   Network
 opensky-network   -                             currently online
 opensky-network   -                       88067 [100%] seconds online (overall)
 opensky-network   -                           5 disconnections
 opensky-network   -                           6 connection attempts
 opensky-network   -                     2858500 bytes sent (32.46/s)
 opensky-network   -                          36 bytes received (0.00/s)

Seems like i am getting alot of disconnections any other people having the same issue

ketilmo, 2023-12-26:
Hi @cws82us313,

Thanks for reaching out. Opensky Network had a server outage earlier in December. This might be the reason for the issues you are seeing. I'm currently experiencing instabilities on my end.

See this update from their team: https://x.com/OpenSkyNetwork/status/1731102369247576473?s=20.

Kind regards,
Ketil

ketilmo, 2024-02-05:
Hi @cws82us313,

OpenSky Network should be operational again now. Have you been able to get it working?

ketilmo, 2024-06-02:
Closing this. Feel free to reopen if you are still experiencing problems.

closed 2024-06-02 by ketilmo