ketilmo/balena-ads-b #105: Wingbits: Warning message in the logs

ketilmo, 2024-06-01:
Over the last few weeks, a warning message has been appearing in the WingBits service logs. This is likely due to changes in the Vector service, as it appears on both new and old versions of the WingBits feeder configuration. 

This is the warning message:

```
WARN transform{component_kind="transform" component_id=remapped component_type=remap}: vector::transforms::remap: VRL compilation warning. warnings=
warning[E900]: unused variable `err`
ââ :1:10
â
1 â read_at, err = .now * 1000
â          ^^^ help: use the result of this expression or remove it
â
= this expression has no side-effects
= see language documentation at https://vrl.dev
= try your code in the VRL REPL, learn more at https://vrl.dev/examples
```

The feeding seems to work properly

ketilmo, 2024-06-01:
@shawaj Would you be able to take a look at this? Also, it would be great if you could confirm that the recent updates to Wingbits have been correctly reflected in recent changes to the balena edition.

shawaj, 2024-06-01:
@ketilmo sure I'll take a look.

When you say the vector service, you mean it's been updated to a new version in later images?

ketilmo, 2024-06-01:
Thanks @shawaj! And correct. When I rolled back to an earlier deployment, the warning message disappeared. So it's likely been introduced by a recent Vector service update (via apt).

shawaj, 2024-07-01:
Same issue as mentioned here https://gitlab.com/wingbits/config/-/merge_requests/22

shawaj, 2024-07-04:
Pulling in latest update (i.e. clicking on deploy with balena) should fix this now.

https://gitlab.com/wingbits/config/-/blob/master/vector.yaml?ref_type=heads

shawaj, 2024-07-04:
Actually probably need to bump version. I will sort a PR shortly

ketilmo, 2024-07-05:
Thanks @shawaj, much apprecaited!

shawaj, 2024-07-05:
Looks like they are no longer tagging the release upstream, so latest config update is 0.0.5 but there is no tag. 

@ketilmo are you happy if I just remove the tag based build and use a commit has instead?

For example https://gitlab.com/wingbits/config/-/blob/96cb3ffb4628fbb8a128a02d691ee16bdff7f917/vector.yaml

ketilmo, 2024-07-05:
That sounds like a good approach, @shawaj! Please go ahead.

PR #118 (2024-07-06): wingbits: bump to latest version
Bump to latest config version

Closes: #105
commit 8abd49e3 (2024-07-06): wingbits: bump to latest version (#118)

* wingbits: bump to latest version

Bump to latest config version

Closes: #105

* wingbits: update vector.yaml

Update vector.yaml to latest

* Update Dockerfile.template

* Delete wingbits/vector.yaml
closed 2024-07-06 by ketilmo