ketilmo/balena-ads-b #107: Plane finder setup instructions doesn't work... Get error

shawaj, 2024-06-01:
The instructions describes how to get a sharecode for plane finder but when you click to generate the share code it goes to a balena error page...
[image]

I guess the device is hosted on balena but the public URL has been toggled off?

shawaj, 2024-06-01:
Actually just discovered you can do issue command:

`pfclient`

On command line. And then access the sharecode creation page via port 30053 on a local device. 

@ketilmo shall I just update the docs to reflect that?

ketilmo, 2024-06-02:
That’s very strange. Do you get this error after extracting the contents of the SharecodeGenerator.zip file and punching the receiver details in the web UI? The posting of the details should go straight to the PlaneFinder servers.

That said, if you have found a way to do this directly through the command line, nothing is better. So yes, please share the details. :)

shawaj, 2024-06-02:
The contents of planefinder sharecode generator HTML seems to link on the form to a balena device...

https://github.com/ketilmo/balena-ads-b/blob/2dfc53b9870e27e1447806759d05bb571acc5c0d/planefinder/SharecodeGenerator.html#L76-L100

Cool I'll update the docs to explain how to do it from the device itself.

shawaj, 2024-06-02:
[image]
[image]
[image]
[image]
![pf5](https://github.com/ketilmo/balena-ads-b/assets/3359418/7d3ef684-0918-408f-8029
[bootstrap.min.zip](https://github.com/user-attachments/files/15525709/bootstrap.min.zip)
-072c581b4c12)

Just posting these here for reference

ketilmo, 2024-06-02:
> The contents of planefinder sharecode generator HTML seems to link on the form to a balena device... [...]

I investigated a bit, and believe the form action specified in the source is just a placeholder. The actual call is made using an ajax JavaScript call. It still works for me the old way in Chrome. Anyway – your approach sounds cleaner.

shawaj, 2024-06-02:
Oh weird. I was doing it on my phone initially so perhaps that's the issue... Probably Chrome on the phone being crappy or something and not loading the JS correctly

shawaj, 2024-06-02:
I checked and it's definitely something to do with phone browser not properly loading the CSS and JS from the subfolder from the relative links. The zip file way worked fine on my laptop.

ketilmo, 2024-06-09:
Thanks for checking and confirming, @shawaj! Glad to hear that.

PR #112 (2024-06-09): readme: update planefinder instructions
Update planefinder setup instructions to allow you to do everything via the device rather than using code from the repo or zip folder.

Closes #107
commit a6929234 (2024-06-09): readme: update planefinder instructions (#112)

Update planefinder setup instructions to allow you to do everything via the device rather than using code from the repo or zip folder.

Closes #107
closed 2024-06-09 by ketilmo