ketilmo/balena-ads-b #108: Updating using blue button is not working

winiberto, 2024-06-01:
i'm trying to update to new version but when I use the blue button it's always in "loading mode". 

[image]

ketilmo, 2024-06-02:
Hi @winiberto,

Thanks for reaching out. The build process typically takes a few minutes but might take longer under certain circumstances. If you click the red `0.0.0+rev2` link (visible in your screenshot), you will see a status log of what's happening. You can also try clicking the blue balena button again to trigger a new build.

Good luck! Let us know if it works for you!

shawaj, 2024-06-02:
I just tried this to update to latest and it worked for me :-)

winiberto, 2024-06-02:
Hi @ketilmo @shawaj I tried it about 4 days ago, then tried again yesterday. After two hours I decided to delete the two attempts and try again. It worked. Thank  you for the support.

shawaj, 2024-06-02:
@winiberto I have found with balena, sometimes it can get stuck in the build process but it never times out. I think it's a bug in their builder. It only seems to happen periodically but still very annoying when it does!

ketilmo, 2024-06-02:
Thanks for the update, @winiberto! Happy feeding. Don't hesitate to reach out if you have any further questions.

closed 2024-06-02 by ketilmo