ketilmo/balena-ads-b #109: [FEATURE REQUEST] Relay station support

ketilmo, 2024-06-02:
In some circumstances, feeding balena-ads-b from an upstream feeder rather than an RTLSDR stick can be beneficial. This includes when a third party feeder is already operational, but you want to feed data to the different services that balena-ads-b supports out of the box.

Quoting [Mhayar](https://forums.balena.io/u/mah/) on the [balena forums](https://forums.balena.io/t/the-balena-ads-b-thread/272290/92?u=ketil):

> I have just got into your nice ads-b work through balena. I have put it on a Raspberry 3b+ and tested with a rtl-sdr dongle that worked very nice; the problem is just the signal inside my building is not so good.
> I have one receiver on rooftop that is working for years and feeding both raw and decrypted outputs to my local virtual radar through LAN.
> 
> Now I want to know if there is any way to make the balena ADS-B station behave somehow like a relay station to get the raw feed from my first unit and share it to other flight trackers? (Which guess it could be done) and if yes, how could it be done (may be inserting variables and values to the console?)

ketilmo, 2024-06-02:
Would it be possible to incorporate this feature request into your work with #104, @shawaj? It seems related in nature.

shawaj, 2024-06-02:
@ketilmo so the concept is to have one balena-ads-b device (or other device) with the rtl-sdr attached over USB, and then another balena-ads-b device receiving data over the local network (or some arbitrary IP/host) from the 1st device?

ketilmo, 2024-06-02:
@shawaj Correct. [Mhayar](https://forums.balena.io/u/mah/) [mentions](https://forums.balena.io/t/the-balena-ads-b-thread/272290/92?u=ketil) that he has a legacy feeder in his attic that he would like to use as an input for the balena device. I believe it would be possible with something like --net-connect, but I haven't been digging very deep so far.

shawaj, 2024-06-02:
Ah okay I'll take a look and reply on the forum post with any questions :-)

ketilmo, 2024-06-02:
Great, thanks @shawaj!