ketilmo/balena-ads-b #110: RadarBox Outdated/New Update??

SkyDiscovery, 2024-06-02:
When you open your RadarBox Station, a pop-up window will appear with the following message:

`A new upgrade is available for your Raspberry Pi RadarBox software!

To upgrade, please enter the following commands on your Raspberry Pi terminal:

sudo apt-get update
sudo apt-get install rbfeeder
sudo systemctl restart rbfeeder
During the upgrade, it will ask you if you want to keep your current version of the "rbfeeder.ini" file. You should press Enter to keep the current version of this file.

Thank you for being a RadarBox feeder! `

You may want to take a look at these changes and the possibility of implementing them to get rid of this popup every time you open your feeder on RadarBox.

shawaj, 2024-06-02:
Where do you see that? 

As I don't get that...
[image]

My guess is you were running an outdated release of balena-ads-b.

You should also be able to update using using the "deploy with balena" button here...
https://github.com/ketilmo/balena-ads-b?tab=readme-ov-file#part-14--updating-to-the-latest-version

That updates all the containers to the latest versions.

SkyDiscovery, 2024-06-02:
Yes, I'm running it right now (last update 7 months ago), rather confusing as the latest release on GitHub was 2022.

> Where do you see that?
> 
> As I don't get that... ![Screenshot_20240602_184454_Chrome](https://private-user-images.githubusercontent.com/3359418/335887980-f1a1c7ee-dfd1-4760-a7c2-d272c605869c.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTczNTA3MDgsIm5iZiI6MTcxNzM1MDQwOCwicGF0aCI6Ii8zMzU5NDE4LzMzNTg4Nzk4MC1mMWExYzdlZS1kZmQxLTQ3NjAtYTdjMi1kMjcyYzYwNTg2OWMuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDJUMTc0NjQ4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGQ5OTE0NTBhMTg5MDU3Mjc4ZGI5MjRlODZiNGNhNjAxZGYyMDcxMTYwNzNkNzBlZjE3OGM3MDI4MGUxMjljNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.cTnFR_ms3SUu93YRr3iPWRmO0wjqBFrXN257VUFNuPs)
> 
> My guess is you were running an outdated release of balena-ads-b.
> 
> You should also be able to update using using the "deploy with balena" button here... https://github.com/ketilmo/balena-ads-b?tab=readme-ov-file#part-14--updating-to-the-latest-version
> 
> That updates all the containers to the latest versions.

SkyDiscovery, 2024-06-02:
maybe show the latest update in the README file?

shawaj, 2024-06-02:
@SkyDiscovery yes I think the repo doesn't use the official GitHub releases anymore and has just been via the commits to the repo essentially

ketilmo, 2024-06-02:
Hi @SkyDiscovery,

@shawaj is right – there haven't been any GitHub release tags recently. Earlier, I used to do a full compatibility check on my test bed of different devices before shrinkwrapping a release. I understand that is a bit confusing.

I have added a feature request to the backlog to remember to look into ways we can make this more intuitive in the future: #111 

Let us know if you were able to update RadarBox.

shawaj, 2024-06-02:
@SkyDiscovery it actually is possible to auto update using a clone of the repo and GitHub actions.

You could set it to check every week or every month for an update and push if there's a change.

I was planning to document this at some point but haven't got around to it yet.

closed 2024-06-08 by SkyDiscovery

ketilmo, 2024-06-09:
@SkyDiscovery @shawaj

Great suggestion about using GitHub actions to automatically updating to the latest version of balena-ads-b. I have added #113 to the backlog.