ketilmo/balena-ads-b #111: [Feature Request] Notify users when a new balena-ads-b release is available

ketilmo, 2024-06-02:
It's a bit hard for users to know when a new version is available. It would be nice to automatically notify them when an update is pending. This can be done by periodically writing a notification to the console log or adding a tag to the balena dashboard. And maybe other ways? User input is appreciated.

shawaj, 2024-06-02:
I think you can subscribe to GitHub releases so using that again could perhaps work? 

Not sure if there's a better way though and requires people to manually subscribe I think.

ketilmo, 2024-06-09:
Good suggestion, @shawaj! That sounds like the most straightforward way to do this. I will also look into implementing a periodic check from the application to the repo, seeing if the installed version is outdated and printing that out in the console or posting it as a tag to the dashboard.