ketilmo/balena-ads-b #116: Can't send data anymore

RudiZhang, 2024-07-03:
Since yesterday, no data has been transmitted. I have no idea what's going on. Log file below. Hope someone can help me!

[balena-ads-b_little-brook-2024-07-03t12_58_51+02_00.txt](https://github.com/user-attachments/files/16082643/balena-ads-b_little-brook-2024-07-03t12_58_51%2B02_00.txt)

ketilmo, 2024-07-05:
Hi @RudiZhang,

I'm sorry to hear that you are experiencing issues. Thanks for submitting the logs. What hardware are you running on, and how long were you running successfully before the issues appeared?

RudiZhang, 2024-07-07:
> Hi @RudiZhang,
> 
> I'm sorry to hear that you are experiencing issues. Thanks for submitting the logs. What hardware are you running on, and how long were you running successfully before the issues appeared?

Hi!

Raspberry Pi 4 
since February didn't have any issue like this

ketilmo, 2024-07-08:
Hi @RudiZhang 

Thanks for clarifying. The problem is that the service dump1090-fa does not find the RTL-SDR radio. 

As a quick first check, try to power the device off completely, unplug the USB radio, plug it in a different USB port, and then power the device on again.

If that doesn't work: Are you running several radios on the same device (for UAT), or just one? Have you set the serial number on the device, or not changed anything from the defaults?

RudiZhang, 2024-07-08:
> Hi @RudiZhang
>
> Thanks for clarifying. The problem is that the service dump1090-fa does not find the RTL-SDR radio.
>
> As a quick first check, try to power the device off completely, unplug the USB [...]

Thanks a lot!! It worked! After a full power-off restart, it finally works again!

closed 2024-07-08 by ketilmo