ketilmo/balena-ads-b #119: Disable wifi-connect service

janipewter, 2024-07-08:
Hi, is there any way to disable the wifi-connect service? I deployed my device originally using wifi but switched to a wired connection as the former was unreliable. Now the device is constantly broadcasting the BalenaWiFi SSID and I want to prevent that. I can't easily redeploy Balena because the device is in a sealed enclosure at the top of a long pole on the roof of a building.

Thanks

ketilmo, 2024-07-08:
Hi @janipewter,

Thanks for reaching out.

In earlier versions of wifi-connect, the hotspot would only start if there was no WiFi _or_ cabled internet connection available. That behaviour seems to have changed lately, however. Now, the hotspot might start even with an active ethernet connection. I'm not sure if this is intentional or not.

If you don't need WiFi, turning off the WiFi radio entirely might be the easiest solution. Here are the steps to do so for Raspberry Pis:

1. Head into the balena dashboard, navigate to the desired fleet, and finally to the device you want to turn off WiFi. 

2. In the menu on the left side of the screen, click "Device Configuration."

3. Locate the setting "Define DT overlays" in the list.

4. Add the following value to the setting, in straight quotes: "disable-wifi". If there already is a value in this field, you should separate the old and new values using a comma, e.g., "vc4-kms-v3d","disable-wifi"

5. Click the "Apply all changes"-button. The device should reboot and come up with the WiFi completely disabled.

To re-enable WiFi in the future, reverse this operation.

Let us know if this works for you.

janipewter, 2024-07-08:
Thanks, I have done this and will report back tomorrow to let you know if it was successful. But from running `ifconfig` in the Host OS console I no longer see the WiFi adapter so I assume it has indeed disabled it.

ketilmo, 2024-07-08:
That's promising, @janipewter! 

Thanks for reporting back. Let us know when you have confirmed that it's working.

From the balena forums’ user mani13jha, an [alternative solution](https://forums.balena.io/t/disable-wi-fi-in-balenos/369846/8?u=ketil) surfaced tonight:

1. From your balena device's dashboard, log into the Host OS terminal.
2. Enter the following command on the command line: `nmcli radio wifi off`

This should switch off the WiFi radio and is reportedly persistent after reboots. To switch WiFi on again, issue the following command on the command line: `nmcli radio wifi off`

janipewter, 2024-07-09:
> Thanks, I have done this and will report back tomorrow to let you know if it was successful. But from running `ifconfig` in the Host OS console I no longer see the WiFi adapter so I assume it has indeed disabled it.

Can confirm this works, thanks. I did not try the `nmcli` method.

ketilmo, 2024-07-09:
That's great to hear, @janipewter! Thanks for confirming. I'll add something about this to the ReadMe at some point. Happy feeding!

closed 2024-07-09 by ketilmo