ketilmo/balena-ads-b #126: [Feature Request] Add support for disabling all services

ketilmo, 2024-08-05:
Currently, it's only possible to (cleanly) disable the services maintained by this project. It would be nice to be able to do that to _all_ services, including `kiosk` and `wifi-connect`.

The following balena feature request is related: [Add option to permanently stop/disable containers](https://feature-requests.balena.io/posts/42/add-option-to-permanently-stop-disable-containers)

Teko012, 2024-08-05:
I thought about this as well, it should be fairly easy to implement. I think we just need a super small image with curl, and a short bash script that runs 1 or 2 curl calls, then also shuts itself down.

shawaj, 2024-08-06:
Are those services using balena blocks?

Teko012, 2024-08-06:
Yes, but we build it ourselves

shawaj, 2024-08-06:
If building ourselves, can't we edit a couple files and just leave the others as upstream? 

But maybe a service container better as can then handle #127 as well in same container I guess 🤔

Teko012, 2024-08-06:
I don't think we can easily mix between them. In the kiosk we just add some packages to something they have already built. Probably a service container is better. But changing the restart policy somewhat already solved this problem, if you stop it once manually, it won't restart.

shawaj, 2024-08-06:
Could just add a small start.sh script here... https://github.com/ketilmo/balena-ads-b/blob/master/kiosk/Dockerfile.template

Which just has the ingestion of disabled services env variable and the curl command to disable? And then just does the standard start procedure? 

But that's assuming not going down the service container route which is likely better I guess

Teko012, 2024-08-06:
Technically we could probably replace the wifi as well similarly with their prebuilt docker image, and then do the modification. There is no real reason for us to build the whole thing.

shawaj, 2024-08-06:
It might help to solve #129 as well by doing it that way instead of submodules...

Teko012, 2024-08-06:
Yes, technically with https://github.com/ketilmo/balena-ads-b/pull/130, we should be able to modify both a little bit, and add an option to disable it and quit.

ketilmo, 2024-08-21:
Great discussion! Let me know if you need anything from me here.

PR #137 (2024-08-21): Modify service stop logic
Modify service stop logic.

Fixes #133
Fixes #135
Closes #126
closed 2024-08-21 by ketilmo