ketilmo/balena-ads-b #127: [Feature Request] Add support for auto-updating balenaOS and Supervisor

ketilmo, 2024-08-06:
First requested by @oliverhavrila in #100, having the option to auto-update balenaOs and Supervisor could be helpful. This could, for instance, be handled by a service controller service – potentially combined with this: #126.

Here is some relevant information:
https://forums.balena.io/t/automated-os-and-supervisor-updates/359033
https://github.com/balena-labs-projects/autohupr

shawaj, 2024-09-20:
@Teko012 do you think it is useful to be able to update the Supervisor automatically independently of the OS?

Reason I ask, is that when you update the OS it automatically updates to the supervisor with that OS...I am not sure there is that much benefit to updating the supervisor in between OS updates? Especially for regularly updated devices like raspberry pi.

But perhaps more relevent for less frequently updated devices I guess.

Teko012, 2024-09-20:
> Reason I ask, is that when you update the OS it automatically updates to the supervisor with that OS.

@shawaj Does it? I wasn't aware / never observed it. I guess they outline some cases where it can be useful: https://docs.balena.io/reference/supervisor/supervisor-upgrades/ We would need to extend this script if we would like to do it though.

shawaj, 2024-09-20:
@Teko012 also, if I remember correctly I think that if you update the supervisor, and then the next OS update includes a supervisor version below that of the current supervisor on device (i.e. it would cause a downgrade of the supervisor) then the OS can't be updated. 

I might be remembering incorrectly though I suppose.

Teko012, 2024-09-20:
I think they explain it here: https://docs.balena.io/reference/OS/updates/update-process/