ketilmo/balena-ads-b #128: Add renovate to repo

Teko012, 2024-08-06:
I started working on a renovate configuration for the repo. Can we install renovate here in the meantime? https://docs.renovatebot.com/getting-started/installing-onboarding/

I hit some bumps with one of the configs that I'm not sure yet how to overcome, so WIP: https://github.com/renovatebot/renovate/discussions/30425

But we could automate most of the dependency updates with it.

Teko012, 2024-08-11:
@ketilmo Can you install this? Once you install it, I will open an MR with my adjustments, I made good progress on it.

Teko012, 2024-08-22:
@ketilmo yes, it would be great if you prioritized it. Just follow [this](https://docs.renovatebot.com/getting-started/installing-onboarding/#hosted-githubcom-app) guide, it's only a few steps.

ketilmo, 2024-08-26:
@Teko012 Renovate is now installed. Let me know if you need me to setup or configure anything else.

Teko012, 2024-08-26:
Thank you. I will create a PR soon with a new/better configuration.

closed 2024-08-26 by Teko012