ketilmo/balena-ads-b #129: Deploy with balena broken due to wifi-connect submodule

shawaj, 2024-08-06:
Deploy with balena breaks due to #114

Also, `balena push` does not work...

```
[Info]             Starting build for balena-ads-b, user gh_shawaj
[Info]             Dashboard link: https://dashboard.balena-cloud.com/apps/2085502/devices
[Error]            Could not detect project type: Service wifi-connect: Could not find a Dockerfile for this service
[Error]            Not deploying release.
Remote build failed
```

It seems to be due to submodules setup as far as i can tell

shawaj, 2024-08-06:
Not sure if this is just me seeing this - or whether it is across the board

shawaj, 2024-08-06:
Mentioned on the docs page...

shawaj, 2024-08-06:
```
Note: Currently git submodules are not supported and will not build properly.
```

https://docs.balena.io/learn/deploy/deploy-with-balena-button/.

shawaj, 2024-08-06:
Also, as a side point, having `url = git@github.com:balena-os/wifi-connect.git` in the `.gitmodules` file means that anyone trying to clone will require a configured SSH key for git. I would suggest better to use the non-SSH based cloning so this is not required

Teko012, 2024-08-06:
Balena push actually works on my side, but the deploy with balena indeed seems to be broken.

shawaj, 2024-08-06:
> Balena push actually works on my side, but the deploy with balena indeed seems to be broken.

the balena push thing was due to me not having cloned recursive and/or it being an SSH github link in .gitmodules and me being on a machine currently where I don't have SSH key set up as per normal so get permission denied error when trying recursive clone:

```
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
```

Teko012, 2024-08-06:
The problem is, that contrary to the kiosk image, they haven't built this image in years, and it's abandoned.

Technically, we could mirror [these](https://github.com/balena-os/wifi-connect/blob/master/scripts/start.sh) [two](https://github.com/balena-os/wifi-connect/blob/master/Dockerfile.template) files, and build it ourselves at deploy, but considering, how often these have changed, I think that should be fine. I will create an MR.

ketilmo, 2024-08-07:
Thanks for identifying and correcting this issue, guys! Unfortunately, I didn't test the blue button deployment before reviewing the previous PR. I guess this was why we didn't use a submodule when we implemented this.

closed 2024-08-07 by ketilmo

shawaj, 2024-08-09:
I think maybe Deploy with Balena is broken on balena end. 

Does it work for you now @Teko012 @ketilmo ?

For me it still doesn't, including this project but also from all on https://hub.balena.io/

shawaj, 2024-08-09:
Create a post here - https://forums.balena.io/t/deploy-with-balena-buttons-no-longer-work/371097

shawaj, 2024-08-09:
Deploy with Balena is working again 👍

ketilmo, 2024-08-26:
Thanks for following up and getting this issue resolved, @shawaj. Great work!

PR #130 (2024-08-07): Fork wifi-connect
This should fix #129