ketilmo/balena-ads-b #133: Some containers that are missing required settings still need to keep running

ketilmo, 2024-08-07:
A case I didn't consider when merging #123 is that some services need the user to interact with the service's shell to register an account. To keep it clean and maintainable, I suggest we pause for missing settings and shut down when a service is specifically disabled through `DISABLED_SERVICES`.

Thoughts, @Teko012 and @shawaj?

Teko012, 2024-08-07:
We could remove it from the parts where it's not disabled, is that what you are suggesting?

shawaj, 2024-08-09:
yeah could just remove it for when they aren't explicitly disabled?

or just add a wait period so that when they are not explicitly disabled it will wait like 1 hour or something before disabling? but maybe that is overkill i dunno...and might confuse people.

Teko012, 2024-08-11:
Should fix this too: https://github.com/ketilmo/balena-ads-b/pull/137

PR #137 (2024-08-21): Modify service stop logic
Modify service stop logic.

Fixes #133
Fixes #135
Closes #126
closed 2024-08-21 by ketilmo