ketilmo/balena-ads-b #135: unable to run /add-serial-1090.sh while dump1090-fa is listed under DISABLED_SERVICES

lrehmann, 2024-08-09:
During my attempted configuration of UAT, I am unable to set the SDR device serial numbers.

Once dump1090-fa is disabled, I have been unsuccessful at starting a terminal session to the dump1090-fa service. 

dump1090-fa log repeats:
```
 dump1090-fa  dump1090-fa is manually disabled.
 dump1090-fa  curl: error while loading shared libraries: /lib/aarch64-linux-gnu/libzstd.so.1: invalid ELF header 
```
 
 attempted session repeats:
```
 SSH session disconnected
SSH reconnecting...
Spawning shell...
Error response from daemon: Container 5e4e384893f9fc28771285d03d6eac0bd9b6320710b5168f9a6a01bd9d4b4889 is restarting, wait until the container is running
Error response from daemon: Container 5e4e384893f9fc28771285d03d6eac0bd9b6320710b5168f9a6a01bd9d4b4889 is restarting, wait until the container is running
SSH session disconnected
SSH reconnecting...
Spawning shell...
sh: line 1: F: command not found
Error: No such container: 5e4e384893f9fc28771285d03d6eac0bd9b6320710b5168f9a6a01bd9d4b4889
Error: No such container: 5e4e384893f9fc28771285d03d6eac0bd9b6320710b5168f9a6a01bd9d4b4889
SSH session disconnected
SSH reconnecting...
SSH session disconnected
SSH reconnecting...
SSH session disconnected
SSH reconnecting...
SSH session disconnected
SSH reconnecting...
```

lrehmann, 2024-08-09:
I was able to work around this issue by connecting the devices to my local machine and setting the corresponding serial numbers manually

```
rtl_eeprom -s 00001090
rtl_eeprom -s 00000978
```

However,  dump978-fa doesn't successfully start:
` dump978-fa  /start.sh: line 66:    14 Bus error               /usr/bin/dump978-fa --sdr driver="$DUMP978_DRIVER" --raw-port 0.0.0.0:30978 --json-port 0.0.0.0:30979 --format CS8 --sdr-auto-gain`

shawaj, 2024-08-10:
You can just remove dump1090-fa from disabled services, run the necessary commands and then disable it again.

shawaj, 2024-08-10:
Sorry, ignore my previous comment.

I think you need to wait until #133 is fixed

shawaj, 2024-08-10:
@ketilmo @teko012 I guess for this the best way is to have a different state such as `DUMP1090_ADD_SERIAL` where it doesn't disable the service it just waits for user to interact with it (i.e. as per previous functionality?)

Or maybe there's a better way?

Perhaps in a refactoring further down the line if #126 introduced some kind of service container these configuration things that don't need anything persisting could perhaps move to that configuration container. But that's possibly overkill I'm not sure.

shawaj, 2024-08-10:
@lrehmann are you trying to run with dump1090 and dump978 but with two rtlsdr dongles connected? Or just one?

Teko012, 2024-08-11:
Here is my proposal for a fix: https://github.com/ketilmo/balena-ads-b/pull/137

lrehmann, 2024-08-12:
@shawaj attempting with 2 dongles connected. #137 will seem to fix the issue I had with setting the serial numbers (though I've manually set them outside the box, so this is no longer a problem for me). 

Is there anything helpful I can provide to debug the Bus error (maybe driver error?) -- this occurred once I had the serial number manually set for both devices. Tried swapping the devices (updated the serial numbers on each) thinking it could be hardware related, but I had the same error.

shawaj, 2024-08-12:
@lrehmann try this...

Click on the *Device Variables*-button in the left-hand menu. Add a new variable named `DUMP978_DEVICE` and set its value to `00000978`.

shawaj, 2024-08-18:
@lrehmann any joy with this?

PR #137 (2024-08-21): Modify service stop logic
Modify service stop logic.

Fixes #133
Fixes #135
Closes #126
closed 2024-08-21 by ketilmo