ketilmo/balena-ads-b #140: Keep lost Internet connection

RudiZhang, 2024-08-15:
I don't know why, but lately my Raspberry Pi keeps disconnecting from the WiFi by itself. I haven't done anything. I've restarted the device every time it disconnects. Could someone please help me? Thanks!

shawaj, 2024-08-18:
It's probably best to post on the raspberry pi forum for this... https://forums.raspberrypi.com/

Also maybe best to use Ethernet if you are having issues with reliability with WiFi.

RudiZhang, 2024-08-22:
> It's probably best to post on the raspberry pi forum for this... https://forums.raspberrypi.com/
> 
> Also maybe best to use Ethernet if you are having issues with reliability with WiFi.

Ok I will try thanks

closed 2024-08-22 by RudiZhang

shawaj, 2024-08-22:
The WiFi stuff is also related to balena so they might be able to provide help on their forums actually... https://forums.balena.io/

ketilmo, 2024-08-26:
Let us know how it goes, @RudiZhang. Best of luck in resolving this problem.

RudiZhang, 2024-09-02:
> Let us know how it goes, @RudiZhang—best of luck in resolving this problem.

It seems to be an unstable connection, although I don't know why, because it seems a bit weird to me. Currently it disconnected from the network again after working stably for 8 days.