ketilmo/balena-ads-b #141: Location Issuses

hankalf, 2024-08-20:
I am having issues with the services not wanting to start due to missing lat & long. I have entered the coordinates under the "Location" tab in balenaCloud as well as entered them under Device Variables. 

[image]

[image]

I have looked through the "Issues & Discussions" and it does not appear that anyone else is having this issue. 

Is someone able to assist with on a fix? 

Thank you.

shawaj, 2024-08-20:
@hankalf you need to follow the instructions here... https://github.com/ketilmo/balena-ads-b?tab=readme-ov-file#part-2--setup-balena-and-configure-the-device

hankalf, 2024-08-20:
I followed that guide again, and after a full redeployed image, I was able to get that issue fixed. 

Now I am hit with a new issue on 3 different services. 

1. adsb-exchange - error connecting to service via terminal. 

[image]
[image]

3. piaware - same error as adsb-exchange.

[image]
[image]

5. planefinder - same issue. 

[image]
[image]

shawaj, 2024-08-21:
This issue is related to #135 and the fix for this is in #137 but hasn't been merged yet. 

You can try manually enabling the containers by clicking the play button next to them but not sure exactly how long that will enable for.

ketilmo, 2024-08-21:
Hi guys,

Thanks for following up on this, @shawaj. And sorry for the inconvenience, @hankalf. I will merge the fix as soon as I've had the chance to review it and update this ticket when it's done. You will have to do a new deployment of the code, @hankalf, either through the blue "Deploy to balena"-button or the `balena push` command line tool.

ketilmo, 2024-08-21:
Hi @hankalf,

The fix has now been merged. Please try deploying the fresh code to your device, and let us know how it goes.

hankalf, 2024-08-21:
Hi @ketilmo & @shawaj, 

Thank you both for your input. I really appreciate it! After redeploying the code to the device, I was able to get into the terminal with no issues. I have now successfully got everything up and running. 

I came across this build while troubleshooting too many issues with a raspberry pi radar feeder. This build is everything I was wanting in my project as I am deploying this at a secondary home that has a full view of the sky all around, so remotely monitoring was a must. I will definitely be following the project for any further updates or improvements! 

Thanks again!

closed 2024-08-21 by hankalf