ketilmo/balena-ads-b #142: WiFi Connect Issue

Tissy, 2024-08-22:
Anyone seen or aware of this issue with `wifi-connect` please?

The WiFi AP is broadcast as it should, I appear to be able to connect to the WiFI, however no landing page is displayed which would enable me to select the access point.

This is what am I seeing in the logs...

2024-08-22T13:49:15+01:00  wifi-connect  Starting WiFi Connect
2024-08-22T13:49:15+01:00  wifi-connect  Starting WiFi Connect
2024-08-22T13:49:16+01:00  wifi-connect  [<unknown>:WARN] Undefined device type: 32
2024-08-22T13:49:16+01:00  wifi-connect  WiFi device: wlan0
2024-08-22T13:49:16+01:00  wifi-connect  Access points: ["vodafoneXXXXXX", "DIRECT-1E-HP Laser 178nw", "EE WiFi", "BT-XXXXXX", "PLUSNET-XXXXXX", "SKYXXXXX", "EE WiFi-X"]
2024-08-22T13:49:16+01:00  wifi-connect  Starting access point...
2024-08-22T13:49:17+01:00  wifi-connect  Access point 'AS1-L_WiFi' created
2024-08-22T13:49:17+01:00  wifi-connect  Starting HTTP server on 192.168.42.1:8181

Many thanks :)

shawaj, 2024-08-22:
@Tissy when you are connected to the WiFi AP, what does it show if you access 192.168.42.1:8181 in a browser?

Tissy, 2024-08-22:
Hi @shawaj , thank you for your reply.

When I connect to that IP, the screen is just blank, so nothing is displayed.

shawaj, 2024-08-22:
Ah ok. Maybe it was a result of #130 🤔

Are you using a raspberry pi with onboard WiFi?

Tissy, 2024-08-22:
That looks to be a different issue as all the code is deploying and the service runs, just no portal page.

Yes, RaspberryPi 4 with onboard WiFi.

shawaj, 2024-08-22:
What I meant was, that change moved the WiFi connect to a local implementation rather than upstream. And #114 changed it from kiosk.

This is a new install? Or it's updating a previous one?

For the time being, you can auto-add the WiFi credentials to the balena image using their "Deploy with Balena" dialog... If you toggle the advanced toggle...
[image]

You should then be able to add pre-configured WiFi credentials.

shawaj, 2024-08-22:
Sorry, wrong screenshot... It's the "Add Device" when where you can add WiFi credentials...

[image]

shawaj, 2024-08-22:
@Tissy did you modify the WiFi AP name? As yours seems to be AS1-L_WiFi instead of balenaWiFi?

Anyway, I have managed to reproduce the same issue...
[image]

No idea why it's happening but will poke around

shawaj, 2024-08-22:
Think this is the issue: https://github.com/balena-os/wifi-connect/issues/588#issue-2438471206

Will find a working version and hard-code something

shawaj, 2024-08-22:
It seems v4.4.6 is the latest working version currently... https://github.com/balena-os/wifi-connect/releases/tag/v4.4.6

shawaj, 2024-08-22:
Ok created #143 as a temporary fix until this upstream PR gets pulled in... https://github.com/balena-os/wifi-connect/pull/566

I tested locally and was able to setup wifi and captive portal showed correctly.

Tissy, 2024-08-23:
That's absolutely brilliant @shawaj , I do indeed now have the portal web page running, thank you very much!
Hopefully Balena will merge soon !
Out of interest, once Balena is connected to an AP using the captive portal, is there a way within Balena cloud to tell it to ignore / reset any existing connections please?  For example, whilst testing, I'd like to go through the setup process from the start, but have to keep changing my WiFi SSID for it to "forget" the existing AP.

shawaj, 2024-08-23:
@Tissy you can either go into the "host" container and type `nmcli device disconnect wlan0' and then hit the restart container icon for wifi-connect 

Or you can go to the wifi-connect container and enter `export DBUS_SYSTEM_BUS_ADDRESS=unix:path=/host/run/dbus/system_bus_socket` and then `./wifi-connect` and then you should get the WiFi AP starting up again which you can connect to to reconfigure the WiFi.

commit 05acf78b (2024-08-26): wifi-connect: temporary fix (#143)

* wifi-connect: temporary fix

Temporary fix for #142
- pull in required changes from https://github.com/balena-os/wifi-connect/pull/566
- increase timeout from 120 seconds to 180 seconds

Closes: #142

* fix: dbus address error

Fixes the error `D-Bus error: In D-Bus address, character '"' should have been escaped`
closed 2024-08-26 by ketilmo