ketilmo/balena-ads-b #160: Build fails on kiosk

serviceman77, 2024-09-14:
I’m using the blue button to update my release but it fails on kiosk with the following error. 

[image]

shawaj, 2024-09-14:
Related to #158 I guess. Looking

shawaj, 2024-09-14:
I think this is actually an issue with Balena container registry. 

Because v2.9.0 is correct as per... https://hub.balena.io/blocks/1916869/browser-aarch64

Will create a thread on balena forum

shawaj, 2024-09-14:
Ok I've discovered something really weird.

If we have in the dockerfile:
```
ARG BROWSER_VERSION=2.9.0
FROM bh.cr/balenalabs/browser-%%BALENA_ARCH%%/$BROWSER_VERSION AS base
```

The build fails with the error message: 
```
Error: pull access denied for bh.cr/balenalabs/browser-aarch64/2.9.0, repository does not exist or may require 'docker login': denied : requested access to the resource is denied
```

But if we instead use:
```
FROM bh.cr/balenalabs/browser-%%BALENA_ARCH%%/2.9.0 AS base
```

It builds successfully.

So it appears to not be liking the ARG variable in the FROM line.... Even though it clearly correctly parses it as the error output shows the correct URL for the package `bh.cr/balenalabs/browser aarch64/2.9.0`

Teko012, 2024-09-15:
As a workaround, technically, renovatebot config could be updated to parse out the version from the `FROM`, then we could drop the `ARG` again

shawaj, 2024-09-15:
> As a workaround, technically, renovatebot config could be updated to parse out the version from the `FROM`, then we could drop the `ARG` again

Yeah I think that's probably the best way, at least until balena fix the parsing (if at all)

klutchell, 2024-09-17:
ARGS in the image name should be supported for public `bh.cr` images now. Can you give it another try?

shawaj, 2024-09-18:
Yep will check now

> ARGS in the image name should be supported for public `bh.cr` images now. Can you give it another try?

shawaj, 2024-09-18:
@klutchell yes it's working now. The latest push succeeded https://github.com/ketilmo/balena-ads-b/actions/runs/10911742222

@serviceman77 you should be good to go now

closed 2024-09-18 by shawaj