ketilmo/balena-ads-b #207: Refactor UAT implementation for ADSB Exchange and clones

ketilmo, 2025-01-13:
From @wiedehopf, via Discord:

> https://github.com/ketilmo/balena-ads-b/blob/master/adsb-exchange/start.sh
your UAT logic is only sending to adsbexchange and not to clones (somewhat unfortunate term) unless i'm missing something.
you can just add a net connector to readsb (named feed-adsbx in this case):
--net-connector dump978,30978,uat_in

> And then remove the special handling for adsbexchange and remove the 978 convert script, it's not been necessary for some time.

> Considering this PR: https://github.com/ketilmo/balena-ads-b/pull/190, you could just use ultrafeeder to send data to all those sites including adsbexchange though (docker-tar1090 / ultrafeeder containers are basically the same, might as well go all the way and add the interface with the feeding that has the MLAT results from all sites nicely bundled up)

shawaj, 2025-01-14:
nice...I think we would just need to detect if uat is enabled higher up in the start script and then add the info for uat to the feeder

shawaj, 2025-01-14:
Could do #167 at same time potentially

wiedehopf, 2025-01-14:
The functionality of feeding adsbexchange and the other sites is integrated with ultrafeeder.
Which is really the same size as just docker-tar1090.

So yeah best to just add ultrafeeder and use it to feed 978 / 1090 / MLAT to the sites.
ultrafeeder will have a separate readsb running (mlathub) which aggregates mlat results.
You can push them into dump1090-fa easily for display there.
They'll be also displayed in the tar1090 webinterface of ultrafeeder.

Should be very possible to just produce a yml to run ultrafeeder from the config.
But that's up to you.

Having --net-connector dump978,30978,uat_in added to the commandline before feed-adsbx (readsb) is started will do the job.
Though i suppose you want to use your env vars.

wiedehopf, 2025-01-14:
If you ignore the RTLSDR part, this is good example for an ultrafeeder yml: https://sdr-enthusiasts.gitbook.io/ads-b/foundations/deploy-ultrafeeder-container

shawaj, 2025-03-24:
@wiedehopf can readsb directly handle an rtl-sdr on 978MHz? Or you have to use dump978 and feed to readsb via a --net-connector?

From what I understand, you still need to use dump978, but from your comments above I am wondering if maybe I misunderstood it.

wiedehopf, 2025-03-24:
no dump978-fa and then readsb uat_in

shawaj, 2025-04-19:
@wiedehopf is there a difference between `--net-connector dump978,30978,uat_in` and `--net-connector dump978,30978,raw_in`? I saw somewhere else (iirc in docker-tar1090 repo) that for ingesting UAT should use `raw_in`. But here you say `uat_in`. Are they interchangeable in this instance?

wiedehopf, 2025-04-19:
Very much not the same thing.
dump978-fa doesn't output raw.
uat2esnt converts uat to avr.

the ax container is unnecessary and deprecated ... you can just run ultrafeeder.

shawaj, 2025-04-19:
> If you find incorrect instructions please link them.

From here - https://github.com/sdr-enthusiasts/docker-tar1090/blob/main/README.md#readsb_net_connector-syntax

```
NOTE: If you have a UAT dongle and use dump978 to decode this, you should use READSB_NET_CONNECTOR to ingest UAT data from dump978. See example below

    environment:
    ...
      - READSB_NET_CONNECTOR=dump978,37981,raw_in;another-data-aggregator.com,30005,beast_reduce_plus_out
    ...
```

wiedehopf, 2025-04-19:
That's a different port.
See the container manual on what is provided on that port.

shawaj, 2025-04-19:
@wiedehopf sorry - i made a mistake in my original question due to incorrect copy pasting. I get that the ports are different for the two and what it is doing under the hood.

So, to clarify, what I meant to ask was, is there any material difference to `--net-connector dump978,30978,uat_in` and `--net-connector dump978,37981,raw_in`?

It seems like in docker-tar1090 they suggest to use raw_in way, but in the docker-dump978 they say that is deprecated and to use uat_in. And you say to use uat_in here. It seems the only difference is that the raw_in way is using uat2esnt to convert into raw before ingesting into readsb, but that this was before you subsequently added direct ingestion via the uat_in. Is it just that the documentation has not been updated?

There is no other benefit or extra data to using one over the other?

wiedehopf, 2025-04-19:
Slight difference as it marks it with a special timestamp similar to MLAT to signifiy it's been converted from UAT to ADS-R.
So yes use uat_in.
But the data is the same otherwise.

shawaj, 2025-04-20:
awesome - thanks @wiedehopf for the clarification...much appreciated! uat_in it is then. And I will submit a PR to the docker-tar1090 repo to update the docs there to add this as well.