ketilmo/balena-ads-b #208: Add Airplanes.live

Huusky, 2025-01-13:
Hey there! Would you be willing to add [airplanes.live](https://airplanes.live)?

We are a group of aviation enthusiasts dedicated to collecting, storing, and distributing unfiltered Mode S, ADS-B, MLAT, and similar publicly available data related to aircraft locations and operations around the world. We aim to provide a sustainable source of unfiltered data to academics, researchers, journalists, and the general public to further activities of an information populace and free press.

I would be happy to answer any questions if you have any.

Thanks!

shawaj, 2025-01-14:
hey @Huusky ...you will find in the README that we already support airplanes.live :-)

https://github.com/ketilmo/balena-ads-b/blob/master/README.md?plain=1#L4

Huusky, 2025-01-16:
Sorry about that, turns out I can't read.

closed 2025-01-16 by Huusky

shawaj, 2025-01-16:
> Sorry about that, turns out I can't read.

No probs at all! It's not listed in the short description as I think we ran out of characters there but is in the full readme 👌