ketilmo/balena-ads-b #227: Update the rtl-sdr library (build from source)

ketilmo, 2025-01-18:
The rtl-sdr Debian package used by dump1090-fa and dump978 is currently outdated. To support newer hardware such as the RTLSDR Blog v4.0, we should consider building the drivers from the Osmocom repository: https://gitea.osmocom.org/sdr/rtl-sdr.git.

shawaj, 2025-01-18:
AFAIK rtlsdr V4 uses the r282 tuner. I'm not sure it's compatible with the software for other SDRs that use R820/R860 tuner.

Might have to have some configuration variable to choose between one or the other?

ketilmo, 2025-01-18:
Previously, the V4 sticks required a modified rtl-sdr version from the RTLSDR Blog team, likely the one you mentioned above. However, I noticed today that Osmocom’s upstream drivers added V4 support back in May last year.

I have several of the V4 sticks in production and have been using the blog drivers with balena-ads-b to make them work. But with the "official" library now supporting them directly, updating the version to enable this out of the box seems like a no-brainer to get them supported in balena-ads-b directly.

I'll build and verify the Osmocom drivers with my testbed and submit a pull request once confirmed. 😊

schubydoo, 2025-02-01:
I was thinking of upgrading to the new v4 devices, should I wait for this PR first?

PR #229 (2025-03-17): Build RTL-SDR drivers from source
This pull request builds the RTL-SDR library from the Osmocom source rather than using the Debian packages. This change is to support newer hardware, such as the RTL-SDR Blog v4, which is currently not supported by the prebuilt packages. I have tested the changes on the v4, and it works fine now.

For dump978, the SoapyRTLSDR module is built manually as well.

As there is no UAT traffic in my area, testing of this functionality would be appreciated.

I'm unsure if I got the renovate settings correct, @Teko012, so any input would be appreciated. 😊

Closes #227
closed 2025-03-17 by ketilmo