ketilmo/balena-ads-b #228: Add support for Airspy

shawaj, 2025-01-18:
Add support for Airspy SDR.

I've got one here so will try to get it working soon :-)

ketilmo, 2025-01-18:
Awesome, @shawaj! I purchased a discounted Airspy Mini during Black Friday, and I have also been thinking about adding support in balena-ads-b for it.

The implementation approach would likely be similar to the Mode S Beast, with a separate service feeding dump1090-fa, right? This repo seems to be a good starting point: https://github.com/wiedehopf/airspy-conf

steffenthorsen, 2025-02-20:
Hi, I'm also interested in support for the Airspy!

shawaj, 2025-05-23:
Closed by #294

shawaj, 2025-05-23:
@steffenthorsen this is now released. Can you give the airspy support a go following the instructions at https://github.com/ketilmo/balena-ads-b?tab=readme-ov-file#airspy-setup

And report back in a new issue if you find any errors :-)

closed 2025-05-23 by shawaj