ketilmo/balena-ads-b #230: Unable to update with blue Balena button

serviceman77, 2025-01-22:
I receive an error when using the blue button to update. Any ideas what is causing this?

[image]

serviceman77, 2025-01-28:
Checking in to see if there is an update to this. Wingbits is sending their geosigners out. What is the workaround to update? Appreciate any suggestions

shawaj, 2025-01-28:
@serviceman77 unfortunately the fix is out of our control - it is an issue on balena side with their backend..

For now, the only way is to push it manually using the instructions here using the balena cli:

https://github.com/ketilmo/balena-ads-b?tab=readme-ov-file#part-2--setup-balena-and-configure-the-device

serviceman77, 2025-01-28:
Thanks! I'm new to programming like this. I guess it's time to learn. Any pointers on this process using windows?

shawaj, 2025-01-28:
I can't give many tips for windows as I've never done this via windows I'm afraid. Only via Linux and Mac. But the following should work.

Download the balenaCLI from here: 

https://github.com/balena-io/balena-cli/blob/master/INSTALL-WINDOWS.md

Then download the zip file of the repo from the main page:

[image]

Then unzip it, and from the CMD prompt, go into that folder and run the `balena login` command and then `balena push my user/myfleet`

Ref:
https://docs.balena.io/learn/getting-started/raspberrypi4-64/python/#install-the-balena-cli
https://docs.balena.io/learn/getting-started/raspberrypi4-64/python/#create-a-release

serviceman77, 2025-01-29:
@shawaj 

Amazing directions! Thank you very much for the assist.

jayscovill, 2025-01-30:
Sorry to piggyback on this but just to clarify, is an update to Balena/Wingbits required for the Geosigner to be recognized and used?

How do we check if it's being found? I've been trying to follow the [instructions](https://docs.wingbits.com/get-started/byod-stations/installing-the-client) on wingbits.com to test with the wingbits binary but I tried opening a shell session to the Wingbits container in the Balena panel but can't find it anywhere.

shawaj, 2025-01-30:
@jayscovill yes that's correct you need to update to the latest version to pick up the new Wingbits client (unless you have updated already within the last 2 weeks)

The binary is either at `/etc/wingbits/wingbits` or `/usr/local/bin/wingbits`

You can run the command `/etc/wingbits/wingbits geosigner info` or on the latest updated version just `wingbits geosigner info` to see if it's working

I'll add this to the readme soon

jayscovill, 2025-01-30:
Thanks for the quick reply. 

I have the variables set for autoupdate so may have the latest version already:

HUP_CHECK_INTERVAL=1d
HUP_TARGET_VERSION=latest/recommended

but:

All I have in /etc/wingbits is a file called "version".

/usr/local/bin/wingbits doesn't existing

Is git clone git@github.com:ketilmo/balena-ads-b.git still the repository we should be using with `balena push` to update manually?  I get a permission denied trying to clone it and the last update shows Dec 2022.

shawaj, 2025-01-30:
@jayscovill that auto update stuff relates to the HostOS. It will not update the application containers themselves, just the version of the balenaOS that runs on the device.

Try `git clone https://github.com/ketilmo/balena-ads-b` as I don't think

jayscovill, 2025-01-31:
Excellent, `git clone https://github.com/ketilmo/balena-ads-b` worked!  Everything updated, thanks.

But:

When I run:

```
root@8fa7011f0bad:/tmp# wingbits version
Version: a2346ea9
Build time: 2025-01-29 12:48:20 UTC
```

the output looks ok but:

```
root@8fa7011f0bad:/tmp# wingbits status
2025/01/31 00:02:29 Failed to display status: failed to get feeder status: failed to get feeder status: exec: "systemctl": executable file not found in $PATH
```

If I run:

```
root@8fa7011f0bad:/tmp# wingbits geosigner info
GeoSigner ID: 012317592FFDBE9BEE
GeoSigner Version: geosigner-0.0.0
Location data: Latitude: 43.505427, Longitude: -80.504324, Satellites: 25
✗ GeoSigner needs full update run `wingbits geosigner flash`
```

it seems to see it but not sure it's actually being used.  And if I try to `wingbits geosigner flash` it fails spectacularly...not sure if that is supposed to work in this dockerized implementation though?

shawaj, 2025-01-31:
@jayscovill wingbits status will not work correctly because balena doesn't use systemctl in the containers. 

You don't need to do the flash thing. It will happen automatically. Just reboot the Pi and leave me GeoSigner plugged in for half an hour or so and it'll do it.

jayscovill, 2025-01-31:
Thanks again for your help!

shawaj, 2025-01-31:
Just FYI @jayscovill I redacted the GeoSigner serial and your lat/lon from the GPS just for security. Probably doesn't matter to much but just in case 🙏

shawaj, 2025-01-31:
@jayscovill actually I'm not sure why `wingbits geosigner flash` didn't work for you.

Unless perhaps it was already running. I don't suppose you have the error output you saw? 

And if you run `wingbits geosigner info` now what does it say for the GeoSigner version? Still 0.0.0 or does it now say 1.0.1?

Thanks

jayscovill, 2025-01-31:
I rebooted the Pi and it's trying to install the firmware update but failing with the same error as when I tried manually.

It just repeats the update loop over and over again with the below error at the end.  Version still shows 0.0.0.0.

```
wingbits  [wingbits-feeder]     Performing full geosigner update...
 wingbits  [wingbits-feeder]     Downloading firmware...
 wingbits  [wingbits-feeder]     Creating virtual environment...
 wingbits  [wingbits-feeder]     Installing esptool...
 wingbits  [wingbits-feeder]     Collecting esptool
 wingbits  [wingbits-feeder]       Using cached esptool-4.8.1.tar.gz (409 kB)
 wingbits  [wingbits-feeder]       Installing build dependencies: started
 wingbits  [wingbits-feeder]       Installing build dependencies: finished with status 'done'
 wingbits  [wingbits-feeder]       Getting requirements to build wheel: started
 wingbits  [wingbits-feeder]       Getting requirements to build wheel: finished with status 'done'
 wingbits  [wingbits-feeder]       Preparing metadata (pyproject.toml): started
 wingbits  [wingbits-feeder]       Preparing metadata (pyproject.toml): finished with status 'done'
 wingbits  [wingbits-feeder]     Collecting bitstring!=4.2.0,>=3.1.6
 wingbits  [wingbits-feeder]       Using cached bitstring-4.3.0-py3-none-any.whl (71 kB)
 wingbits  [wingbits-feeder]     Collecting cryptography>=2.1.4
 wingbits  [wingbits-feeder]       Using cached cryptography-44.0.0.tar.gz (710 kB)
 wingbits  [wingbits-feeder]       Installing build dependencies: started
 wingbits  [wingbits-feeder]       Installing build dependencies: finished with status 'error'
 wingbits  [wingbits-feeder]       error: subprocess-exited-with-error
 wingbits  [wingbits-feeder]       
 wingbits  [wingbits-feeder]       Ã pip subprocess to install build dependencies did not run successfully.
 wingbits  [wingbits-feeder]       â exit code: 1
 wingbits  [wingbits-feeder]       â°â> [72 lines of output]
 wingbits  [wingbits-feeder]           Collecting maturin<2,>=1
 wingbits  [wingbits-feeder]             Using cached maturin-1.8.1-py3-none-manylinux_2_17_armv7l.manylinux2014_armv7l.musllinux_1_1_armv7l.whl (7.5 MB)
 wingbits  [wingbits-feeder]           Collecting cffi>=1.12
 wingbits  [wingbits-feeder]             Using cached cffi-1.17.1.tar.gz (516 kB)
 wingbits  [wingbits-feeder]             Installing build dependencies: started
 wingbits  [wingbits-feeder]             Installing build dependencies: finished with status 'done'
 wingbits  [wingbits-feeder]             Getting requirements to build wheel: started
 wingbits  [wingbits-feeder]             Getting requirements to build wheel: finished with status 'done'
 wingbits  [wingbits-feeder]             Preparing metadata (pyproject.toml): started
 wingbits  [wingbits-feeder]             Preparing metadata (pyproject.toml): finished with status 'done'
 wingbits  [wingbits-feeder]           Collecting setuptools!=74.0.0,!=74.1.0,!=74.1.1,!=74.1.2
 wingbits  [wingbits-feeder]             Using cached setuptools-75.8.0-py3-none-any.whl (1.2 MB)
 wingbits  [wingbits-feeder]           Collecting pycparser
 wingbits  [wingbits-feeder]             Using cached pycparser-2.22-py3-none-any.whl (117 kB)
 wingbits  [wingbits-feeder]           Building wheels for collected packages: cffi
 wingbits  [wingbits-feeder]             Building wheel for cffi (pyproject.toml): started
 wingbits  [wingbits-feeder]             Building wheel for cffi (pyproject.toml): finished with status 'error'
 wingbits  [wingbits-feeder]             error: subprocess-exited-with-error
 wingbits  [wingbits-feeder]           
 wingbits  [wingbits-feeder]             Ã Building wheel for cffi (pyproject.toml) did not run successfully.
 wingbits  [wingbits-feeder]             â exit code: 1
 wingbits  [wingbits-feeder]             â°â> [44 lines of output]
 wingbits  [wingbits-feeder]           
 wingbits  [wingbits-feeder]                     No working compiler found, or bogus compiler options passed to
 wingbits  [wingbits-feeder]                     the compiler from Python's standard "distutils" module.  See
 wingbits  [wingbits-feeder]                     the error messages above.  Likely, the problem is not related
 wingbits  [wingbits-feeder]                     to CFFI but generic to the setup.py of any Python package that
 wingbits  [wingbits-feeder]                     tries to compile C code.  (Hints: on OS/X 10.8, for errors about
 wingbits  [wingbits-feeder]                     -mno-fused-madd see http://stackoverflow.com/questions/22313407/
 wingbits  [wingbits-feeder]                     Otherwise, see https://wiki.python.org/moin/CompLangPython or
 wingbits  [wingbits-feeder]                     the IRC channel #python on irc.libera.chat.)
 wingbits  [wingbits-feeder]           
 wingbits  [wingbits-feeder]                     Trying to continue anyway.  If you are trying to install CFFI from
 wingbits  [wingbits-feeder]                     a build done in a different context, you can ignore this warning.
 wingbits  [wingbits-feeder]           
 wingbits  [wingbits-feeder]                 running bdist_wheel
 wingbits  [wingbits-feeder]                 running build
 wingbits  [wingbits-feeder]                 running build_py
 wingbits  [wingbits-feeder]                 creating build/lib.linux-armv7l-cpython-311/cffi
 wingbits  [wingbits-feeder]                 copying src/cffi/setuptools_ext.py -> build/lib.linux-armv7l-cpython-311/cffi
[... cut ...]
 wingbits  [wingbits-feeder]       note: This error originates from a subprocess, and is likely not a problem with pip.
 wingbits  [wingbits-feeder]     error: subprocess-exited-with-error
 wingbits  [wingbits-feeder]     Ã pip subprocess to install build dependencies did not run successfully.
 wingbits  [wingbits-feeder]     â exit code: 1
 wingbits  [wingbits-feeder]     â°â> See above for output.
 wingbits  [wingbits-feeder]     note: This error originates from a subprocess, and is likely not a problem with pip.
```

shawaj, 2025-01-31:
@jayscovill very strange! 

Which version of raspberry pi is this on? Is it a raspberry pi 2 or 3?

jayscovill, 2025-01-31:
Pi 3.

shawaj, 2025-01-31:
> Pi 3.

From the error output, I'm guessing you are using a 32 bit version of the OS as well @jayscovill ? Not the Raspberry Pi 3 (64 bit OS) option in balena? 

I think basically it's missing the python wheels for this architecture, and it is also missing a compiler for C code which means it can't then build the package locally itself. 

If you go to the Wingbits container terminal and issue this command `install_packages gcc` it hopefully then will be able to proceed with the update once that installation finishes.

Otherwise I'll have to do some more digging.

shawaj, 2025-01-31:
Actually @jayscovill if you haven't already done the GCC thing above, you might be better to try this first: 

```
pip install cryptography --extra-index-url=https://www.piwheels.org/simple
```

jayscovill, 2025-01-31:
It appears indeed that I'm running a 32bit OS:

```
root@f9ce1e6:~# uname -m
armv7l
```

Running `install_packages gcc` succeeds but still results in the same error upgrading the firmware.

And:

```
root@8fa7011f0bad:/tmp# pip install cryptography --extra-index-url=https://www.piwheels.org/simple
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

shawaj, 2025-01-31:
@jayscovill try just this then: 
```
export PIP_EXTRA_INDEX_URL=https://www.piwheels.org/simple
```

Otherwise I'm likely going to have to add something to the dockerfile in an update.

jayscovill, 2025-01-31:
Sorry, try `export PIP_EXTRA_INDEX_URL=https://www.piwheels.org/simple` and then what? The firmware upgrade?

 Results in the similar error but some different info as well:

```
root@8fa7011f0bad:/tmp# wingbits geosigner flash
Performing full geosigner update...
Downloading firmware...
Creating virtual environment...
Installing esptool...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting esptool
  Downloading https://www.piwheels.org/simple/esptool/esptool-4.8.1-py3-none-any.whl (534 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 534.2/534.2 kB 80.3 kB/s eta 0:00:00
Collecting bitstring!=4.2.0,>=3.1.6
  Downloading https://www.piwheels.org/simple/bitstring/bitstring-4.3.0-py3-none-any.whl (71 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 71.9/71.9 kB 13.8 kB/s eta 0:00:00
Collecting cryptography>=2.1.4
  Downloading https://www.piwheels.org/simple/cryptography/cryptography-44.0.0-cp37-abi3-linux_armv7l.whl (1.6 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.6/1.6 MB 1.7 MB/s eta 0:00:00
Collecting ecdsa>=0.16.0
  Downloading https://www.piwheels.org/simple/ecdsa/ecdsa-0.19.0-py2.py3-none-any.whl (146 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 146.6/146.6 kB 1.2 MB/s eta 0:00:00
Collecting pyserial>=3.3
  Downloading https://www.piwheels.org/simple/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 90.6/90.6 kB 1.6 MB/s eta 0:00:00
Collecting reedsolo<1.8,>=1.5.3
  Downloading https://www.piwheels.org/simple/reedsolo/reedsolo-1.7.0-py3-none-any.whl (32 kB)
Collecting PyYAML>=5.1
  Downloading https://www.piwheels.org/simple/pyyaml/PyYAML-6.0.2-cp311-cp311-linux_armv7l.whl (45 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 45.4/45.4 kB 199.8 kB/s eta 0:00:00
Collecting intelhex
  Downloading https://www.piwheels.org/simple/intelhex/intelhex-2.3.0-py2.py3-none-any.whl (50 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 50.9/50.9 kB 860.8 kB/s eta 0:00:00
Collecting argcomplete>=3
  Downloading https://www.piwheels.org/simple/argcomplete/argcomplete-3.5.3-py3-none-any.whl (43 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 43.6/43.6 kB 281.9 kB/s eta 0:00:00
Collecting bitarray<3.1,>=3.0.0
  Downloading bitarray-3.0.0.tar.gz (126 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 126.6/126.6 kB 573.8 kB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Collecting cffi>=1.12
  Downloading https://www.piwheels.org/simple/cffi/cffi-1.17.1-cp311-cp311-linux_armv7l.whl (384 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 384.5/384.5 kB 110.1 kB/s eta 0:00:00
Collecting six>=1.9.0
  Downloading https://www.piwheels.org/simple/six/six-1.17.0-py2.py3-none-any.whl (11 kB)
Collecting pycparser
  Downloading https://www.piwheels.org/simple/pycparser/pycparser-2.22-py3-none-any.whl (117 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 117.6/117.6 kB 42.9 kB/s eta 0:00:00
Installing collected packages: reedsolo, pyserial, intelhex, bitarray, six, PyYAML, pycparser, bitstring, argcomplete, ecdsa, cffi, cryptography, esptool
  DEPRECATION: bitarray is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559
  Running setup.py install for bitarray ... error
  error: subprocess-exited-with-error
  
  × Running setup.py install for bitarray did not run successfully.
  │ exit code: 1
  ╰─> [28 lines of output]
      running install
      /tmp/wingbits_flash/.venv/lib/python3.11/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      running build
      running build_py
      creating build
      creating build/lib.linux-armv7l-cpython-311
      creating build/lib.linux-armv7l-cpython-311/bitarray
      copying bitarray/test_bitarray.py -> build/lib.linux-armv7l-cpython-311/bitarray
      copying bitarray/test_util.py -> build/lib.linux-armv7l-cpython-311/bitarray
      copying bitarray/__init__.py -> build/lib.linux-armv7l-cpython-311/bitarray
      copying bitarray/util.py -> build/lib.linux-armv7l-cpython-311/bitarray
      copying bitarray/bitarray.h -> build/lib.linux-armv7l-cpython-311/bitarray
[... 4 lines cut ...]
      copying bitarray/__init__.pyi -> build/lib.linux-armv7l-cpython-311/bitarray
      running build_ext
      building 'bitarray._bitarray' extension
      creating build/temp.linux-armv7l-cpython-311
      creating build/temp.linux-armv7l-cpython-311/bitarray
      arm-linux-gnueabihf-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/tmp/wingbits_flash/.venv/include -I/usr/include/python3.11 -c bitarray/_bitarray.c -o build/temp.linux-armv7l-cpython-311/bitarray/_bitarray.o
      bitarray/_bitarray.c:12:10: fatal error: Python.h: No such file or directory
         12 | #include "Python.h"
            |          ^~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> bitarray

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
Error: failed to install esptool: exit status 1
```

Again, appreciate the effort to help me out.

shawaj, 2025-01-31:
@jayscovill getting somewhere as it's found the cryptography package now at least! 

Next thing to try: 
```
install_packages python3-bitarray python3-cryptography
```

jayscovill, 2025-01-31:
That seemed to go off without a hitch:

```
root@8fa7011f0bad:/tmp# install_packages python3-bitarray python3-cryptography
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  python3-cffi-backend
Suggested packages:
  python-cryptography-doc python3-cryptography-vectors
The following NEW packages will be installed:
  python3-bitarray python3-cffi-backend python3-cryptography
0 upgraded, 3 newly installed, 0 to remove and 12 not upgraded.
Need to get 813 kB of archives.
After this operation, 3422 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main armhf python3-bitarray armhf 2.7.3-1 [138 kB]
Get:2 http://deb.debian.org/debian bookworm/main armhf python3-cffi-backend armhf 1.15.1-5+b1 [75.1 kB]
Get:3 http://deb.debian.org/debian bookworm/main armhf python3-cryptography armhf 38.0.4-3+deb12u1 [600 kB]
Fetched 813 kB in 0s (1771 kB/s)              
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package python3-bitarray.
(Reading database ... 12030 files and directories currently installed.)
Preparing to unpack .../python3-bitarray_2.7.3-1_armhf.deb ...
Unpacking python3-bitarray (2.7.3-1) ...
Selecting previously unselected package python3-cffi-backend:armhf.
Preparing to unpack .../python3-cffi-backend_1.15.1-5+b1_armhf.deb ...
Unpacking python3-cffi-backend:armhf (1.15.1-5+b1) ...
Selecting previously unselected package python3-cryptography.
Preparing to unpack .../python3-cryptography_38.0.4-3+deb12u1_armhf.deb ...
Unpacking python3-cryptography (38.0.4-3+deb12u1) ...
Setting up python3-bitarray (2.7.3-1) ...
Setting up python3-cffi-backend:armhf (1.15.1-5+b1) ...
Setting up python3-cryptography (38.0.4-3+deb12u1) ...
```

But:

```
root@8fa7011f0bad:/tmp# wingbits geosigner flash
Performing full geosigner update...
Downloading firmware...
Creating virtual environment...
Installing esptool...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting esptool
  Using cached https://www.piwheels.org/simple/esptool/esptool-4.8.1-py3-none-any.whl (534 kB)
Collecting bitstring!=4.2.0,>=3.1.6
  Using cached https://www.piwheels.org/simple/bitstring/bitstring-4.3.0-py3-none-any.whl (71 kB)
Collecting cryptography>=2.1.4
  Using cached https://www.piwheels.org/simple/cryptography/cryptography-44.0.0-cp37-abi3-linux_armv7l.whl (1.6 MB)
Collecting ecdsa>=0.16.0
  Using cached https://www.piwheels.org/simple/ecdsa/ecdsa-0.19.0-py2.py3-none-any.whl (146 kB)
Collecting pyserial>=3.3
  Using cached https://www.piwheels.org/simple/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting reedsolo<1.8,>=1.5.3
  Using cached https://www.piwheels.org/simple/reedsolo/reedsolo-1.7.0-py3-none-any.whl (32 kB)
Collecting PyYAML>=5.1
  Using cached https://www.piwheels.org/simple/pyyaml/PyYAML-6.0.2-cp311-cp311-linux_armv7l.whl (45 kB)
Collecting intelhex
  Using cached https://www.piwheels.org/simple/intelhex/intelhex-2.3.0-py2.py3-none-any.whl (50 kB)
Collecting argcomplete>=3
  Using cached https://www.piwheels.org/simple/argcomplete/argcomplete-3.5.3-py3-none-any.whl (43 kB)
Collecting bitarray<3.1,>=3.0.0
  Using cached bitarray-3.0.0.tar.gz (126 kB)
  Installing build dependencies ... error
  error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 2
  ╰─> [1 lines of output]
      /tmp/wingbits_flash/.venv/bin/python3: can't open file '/tmp/wingbits_flash/.venv/lib/python3.11/site-packages/pip': [Errno 2] No such file or directory
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 2
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
Error: failed to install esptool: exit status 1
```

shawaj, 2025-01-31:
Hmm that's very odd. I'm really not sure where to go from here and it is 2.30 am here so I'm going to go to sleep and come back to this tomorrow. And maybe try to reproduce it locally.

One way that will definitely work to fix it is if you use a new fleet on balena with device type Raspberry Pi 3 (64 bit OS). But obviously that would mean entirely re-flashing your device and setting up all the device variables again and stuff so is a bit of a faff. But if you are eager to get it working ASAP this is probably going to be the quickest way.

jayscovill, 2025-01-31:
Please get some rest :)

If there is no danger of losing my feeder status since it's DIY by rebuilding with a 64-bit OS I'm game for that.  You're right, it'll be a bit of work since it's been so long since I set it up I'll have to re-learn the process.  But it is probably the right long term solution.

shawaj, 2025-01-31:
You will have a little bit of downtime on the station whilst you swap to new fleet (if you use a different SD card to existing one with raspi3 32 bit on it it'll be less - and has the add bonus of meaning you have a backup to revert to just by changing SD card if anything goes wrong!).

But apart from that, you won't lose your ability to do DIY or connect the GeoSigner or anything. 

I'll still try and get this fixed in any case, but I'm away ATM and not sure if I have a raspberry pi 3 that I can access remotely to test with 😂 which is the main reason it'll take a while. 

Anyway let me know if you get stuck on anything and I'll be happy to help more tomorrow

schubydoo, 2025-01-31:
Following because I'm in the exact same scenario as @jayscovill .  Funny enough I was even thinking of re-doing everything in 64 bit, but also its been forever since I originally set it up, so I haven't decided yet.

jayscovill, 2025-01-31:
> Hmm that's very odd. I'm really not sure where to go from here and it is 2.30 am here so I'm going to go to sleep and come back to this tomorrow. And maybe try to reproduce it locally.
>
> One way that [...]

Welp, I've redeployed with a 64-bit OS, set all my env variables and deployed the packages but for some reason most of my services, including the important ones like tar90 and wingbits don't want to start.

tar90 is stuck in "Downloaded" and wingbits is  stuck in "Installed" but won't start with a "Network error".

shawaj, 2025-01-31:
@jayscovill just to check, did you add all of the lat/lon etc variables same as before? 

Try opening a terminal to host OS and issuing the following commands:

```
systemctl stop balena-supervisor
systemctl stop balena
rm -rf /var/lib/balena/{aufs,overlay,containers,image,tmp}
systemctl start balena
update-balena-supervisor
```

If that doesn't work, would you be happy to give me access to the device on balena to poke around with and see if I can find any clues?

Re yours @schubydoo you should be able to just run `install_packages python3-bitarray` however what I don't understand is how it's possible that this works on my Raspberry Pi 4 but not on anyone else's. I'm extremely confused about that 🤔

Will do some more digging once I get home.

jayscovill, 2025-01-31:
GRRR.  I found part of my error.  I needed to define my WINGBITS_DEVICE_ID variable for the "wingbits" service, not "All Services" and wingbits fired up.  For some reason "All Services" worked on the old feeder but didn't.

However it seems like dump1090-fa isn't working correctly and perhaps it's because dump978-fa is also running and is fighting for the RTL-SDR?  I've used "DISABLED_SERVICES" and added fump978-fa but it always wants to start on a reboot and won't let me stop it.

shawaj, 2025-01-31:
> Just to add to this, I actually created a new device in the same old fleet instead of creating a new fleet...didn't seem like a problem doing that so hopefully that has nothing to do with my problems.

Ah that's your issue... It has to be a new fleet with the default device type as something 64 bit (such as raspberry pi 3 64 bit OS). As right now you have the core OS as 64 bit bit it's trying to run 32 bit application containers still. This sometimes is ok but can cause weird issues. You need the fleet device type to be 64 bit so that the application is rebuilt as 64 bit containers.

jayscovill, 2025-01-31:
You're kidding me...gah! I should have listened to you in the first place :P I shall start again.  Thanks!

schubydoo, 2025-01-31:
RE: Geosigner flashing

Ok I started over by restarting the container because I got too much in the weeds.  Default boot shows this error when trying to build **CFFI** (I'll truncate it a bit for room).

```
No working compiler found, or bogus compiler options passed to
                the compiler from Python's standard "distutils" module.  See
                the error messages above.
```

So then I try `install_packages gcc` and rerun the flasher.  Then CFFI complains again:

```
src/c/_cffi_backend.c:2:10: fatal error: Python.h: No such file or directory
                2 | #include <Python.h>
                  |          ^~~~~~~~~~
            compilation terminated.
            error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
            [end of output]

```

So then I try `install_packages python3-dev` and rerun the flasher.   Then another complaint:

```
 src/c/_cffi_backend.c:15:10: fatal error: ffi.h: No such file or directory
               15 | #include <ffi.h>
                  |          ^~~~~~~
            compilation terminated.
            error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
            [end of output]
```

So then I try `install_packages libffi-dev` and rerun the flasher.   And this is where I get stuck.

```
Collecting cryptography>=2.1.4
  Using cached cryptography-44.0.0.tar.gz (710 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      
      Cargo, the Rust package manager, is not installed or is not on PATH.
      This package requires Rust and Cargo to compile extensions. Install it through
      the system's package manager or via https://rustup.rs/
      
      Checking for Rust toolchain....
      [end of output]
```

I tried installing cargo, but the version of rust in the debian bookworm repo was too low (it will spit out that error after compiling for what felt like 20 minutes).  I also tried installing rust using the rustup process, but no matter what changes I make to path, the geosigner flasher can't find Cargo to finish the build for crypography.

jayscovill, 2025-02-01:
Wierd, everything seems to be up now ok.  But we're back to the original problem with a twist.

I saw it trying to update the Geosigner in the logs and it failed but not with the same message as what I was originally getting that triggered this little adventure.

Manually trying to update it I get this now:

```
root@e7c5e9fb738d:/tmp# wingbits geosigner info
Error: no GeoSigner device found
```

shawaj, 2025-02-01:
What error are you getting now when it tries to flash by itself? 

Just FYI wingbits status is still expected not to work because it is using systemctl still under the hood so it will say it's not running even though it actually is. However wingbits geosigner info and wingbits geosigner flash should still work.

Can you filter your logs to just Wingbits container and then click the little replay icon next to the container in the list so that it restarts the container? And then copy over the output including any error messages.

Just to check, the power supply you are using, is it an official raspberry pi one, or one that's known to be good and not under powered? I'm wondering if some weirdness is being caused by undervoltage causing weird issues on USB. Just a thought.

jayscovill, 2025-02-01:
My bad, I remember you saying that already.

The error was: 

`2025-01-31T18:58:08-05:00  wingbits  [wingbits-feeder]     A fatal error occurred: Failed to write to target Flash after seq 134 (result was 01050000: Requested resource not found)`

shawaj, 2025-02-01:
@schubydoo you can try just installing python3-cryptography with `install_packages python3-cryptography` or by adding the environment variable PIP_EXTRA_INDEX_URL=https://www.piwheels.org/simple

It should work, but may then fail further in. I'm still unclear how my pi 4 works fine but yours doesn't. Really strange.

shawaj, 2025-02-01:
> My bad, I remember you saying that already.
> 
> The error was:
> 
> `2025-01-31T18:58:08-05:00 wingbits [wingbits-feeder] A fatal error occurred: Failed to write to target Flash after seq 134 (result was 01050000: Requested resource not found)`

This seems to me like it's losing connection. It could be a power supply issue causing not enough power on the USB bus. Or, if you are using the dongle on the extension cable, you could try it without the extension and with the GeoSigner directly connected to USB. Something doesn't seem to be quite right. Hard to say what exactly to be honest.

jayscovill, 2025-02-01:
> This seems to me like it's losing connection. It could be a power supply issue causing not enough power on the USB bus. Or, if you are using the dongle on the extension cable, you could try it [...]

It's directly connected atm.  Ok, well I'll play around.  Thanks for sticking around with me on all this!

jayscovill, 2025-02-01:
> This seems to me like it's losing connection. It could be a power supply issue causing not enough power on the USB bus. Or, if you are using the dongle on the extension cable, you could try it [...]

Yeah dunno...it's a 5V 3mA power supply...so thinking maybe my Geosigner is borked because of a bad firmware update.  Doesn't seem to be detected at all and status command just hangs now.

schubydoo, 2025-02-01:
@jayscovill can you tell me what commands you ran before the attempted flash? was it:

`export PIP_EXTRA_INDEX_URL=https://www.piwheels.org/simple` and `install_packages python3-cryptography` ? I want to see if I can get as far as you did.

jayscovill, 2025-02-01:
Was just talking with a Wingbits guy on the Discord and he was saying I could try a `wingbits geosigner flash` command again but wingbits can't be running while I do it...which is tricky when the binary is in the wingbits container.

schubydoo, 2025-02-01:
@shawaj I got it to work using this exact command upon restarting the container:

`export PIP_EXTRA_INDEX_URL=https://www.piwheels.org/simple && install_packages python3-bitarray python3-cryptography gcc python3-dev`

jayscovill, 2025-02-01:
> [@jayscovill](https://github.com/jayscovill) can you tell me what commands you ran before the attempted flash? was it:
> 
> `export PIP_EXTRA_INDEX_URL=https://www.piwheels.org/simple` and `install_packages python3-cryptography` ? I want to see if I can get as far as you did.

I'm pretty sure my issue now is that my Geosigner is no longer being recognized as it looks to be failing on connecting to it via USB serial:

```
Serial port /dev/ttyAMA0
Connecting...
/dev/ttyAMA0 failed to connect: [Errno 25] Inappropriate ioctl for device
Serial port /dev/ttyACM0
Connecting.......
Traceback (most recent call last):
  File "/tmp/wingbits_flash/.venv/lib/python3.11/site-packages/esptool/loader.py", line 782, in connect
    self.check_chip_id()
  File "/tmp/wingbits_flash/.venv/lib/python3.11/site-packages/esptool/loader.py", line 1557, in check_chip_id
    chip_id = self.get_chip_id()
```

schubydoo, 2025-02-01:
> I'm pretty sure my issue now is that my Geosigner is no longer being recognized as it looks to be failing on connecting to it via USB serial:
>
> ```
> Serial port /dev/ttyAMA0
> Connecting...
> /dev/ttyAMA0 [...]

Have you tried disconnecting all USB receivers except for the Geosigner? I didn't need to do that, but just something to try.

jayscovill, 2025-02-01:
Ya, no dice...same result :(

shawaj, 2025-02-01:
@schubydoo so yours worked now?

@jayscovill what power supply are you using? I'm still thinking it could be a power / undervoltage issue

jayscovill, 2025-02-01:
Well, tried again and got further this time but it's still crapping out in the during the flashing process.  I have no other devices connected to the Pi so if this is a power problem and it won't work with a 5v 3A p/s then I don't know what else to try.

```
Erasing flash...
WARNING: Security features enabled, so not changing any flash settings.
Took 17.16s to erase flash block
Writing at 0x00030400... (4 %)Traceback (most recent call last):
  File "/tmp/wingbits_flash/.venv/lib/python3.11/site-packages/esptool/__init__.py", line 1314, in _main
    main()
  File "/tmp/wingbits_flash/.venv/lib/python3.11/site-packages/esptool/__init__.py", line 1036, in main
    operation_func(esp, args)
  File "/tmp/wingbits_flash/.venv/lib/python3.11/site-packages/esptool/cmds.py", line 657, in write_flash
    esp.flash_block(block, seq)
  File "/tmp/wingbits_flash/.venv/lib/python3.11/site-packages/esptool/loader.py", line 927, in flash_block
    self.check_command(
  File "/tmp/wingbits_flash/.venv/lib/python3.11/site-packages/esptool/loader.py", line 510, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/wingbits_flash/.venv/lib/python3.11/site-packages/esptool/loader.py", line 479, in command
    p = self.read()
        ^^^^^^^^^^^
  File "/tmp/wingbits_flash/.venv/lib/python3.11/site-packages/esptool/loader.py", line 412, in read
    return next(self._slip_reader)
           ^^^^^^^^^^^^^^^^^^^^^^^
StopIteration

A fatal error occurred: The chip stopped responding.
Error: failed to flash firmware: exit status 2
```

shawaj, 2025-02-01:
5v 3a **should** be ok... But it could still be undervoltage. 

Is it a RasPi official branded power supply or something else? The RasPi ones are actually rated at 5.25 volts because then when with the voltage drop it usually sits at around 5v.

The weirdness and changing nature of the connection issue suggests to me an issue with undervoltage (esp32 can be a bit temperamental with even slight power issues) however it's a bit hard to say.

jayscovill, 2025-02-01:
Not official RasPi.  Tried several variations of different power sources all rated at least 3A but still no dice.  Will hunt around for an official one.

shawaj, 2025-02-01:
> Not official RasPi. Tried several variations of different power sources all rated at least 3A but still no dice. Will hunt around for an official one.

Yeah, if you can find one that's rated at 5.25v or even 5.15v it may solve the issue. My guess is the 5v bus on the USB on the raspberry pi is dropping too low. 

Hard to say for sure though, unless you have a USB power meter to hand?

shawaj, 2025-02-01:
Also try these commands as well, might give some clues https://github.com/ketilmo/balena-ads-b/issues/230#issuecomment-2628989073

jayscovill, 2025-02-01:
```
root@fae08cc:~# lsusb
Bus 001 Device 004: ID 0bda:2838 Realtek RTL2838UHIDIR
Bus 001 Device 005: ID 303a:1001 Espressif USB JTAG/serial debug unit
Bus 001 Device 006: ID 0424:7800  
Bus 001 Device 003: ID 0424:2514  
Bus 001 Device 002: ID 0424:2514  
Bus 001 Device 001: ID 1d6b:0002 Linux 5.15.92-v8 dwc_otg_hcd DWC OTG Controller
```

```
root@fae08cc:~# dmesg | grep usb
[    0.217746] usbcore: registered new interface driver usbfs
[    0.217872] usbcore: registered new interface driver hub
[    0.218021] usbcore: registered new device driver usb
[    0.218587] usb_phy_generic phy: supply vcc not found, using dummy regulator
[    0.219021] usb_phy_generic phy: dummy supplies not allowed for exclusive requests
[    0.927621] dwc_otg 3f980000.usb: DWC OTG Controller
[    0.927679] dwc_otg 3f980000.usb: new USB bus registered, assigned bus number 1
[    0.927795] dwc_otg 3f980000.usb: irq 74, io mem 0x00000000
[    0.928586] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.15
[    0.928622] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.928651] usb usb1: Product: DWC OTG Controller
[    0.928677] usb usb1: Manufacturer: Linux 5.15.92-v8 dwc_otg_hcd
[    0.928732] usb usb1: SerialNumber: 3f980000.usb
[    0.933356] usbcore: registered new interface driver usb-storage
[    0.933566] usbcore: registered new interface driver usbserial_generic
[    0.933633] usbserial: USB Serial support registered for generic
[    0.946160] usbcore: registered new interface driver usbhid
[    0.946171] usbhid: USB HID core driver
[    1.212780] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    1.399263] usb 1-1: New USB device found, idVendor=0424, idProduct=2514, bcdDevice= b.b3
[    1.399299] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    1.691756] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    1.781214] usb 1-1.1: New USB device found, idVendor=0424, idProduct=2514, bcdDevice= b.b3
[    1.781248] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    1.860758] usb 1-1.3: new high-speed USB device number 4 using dwc_otg
[    1.966204] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=2838, bcdDevice= 1.00
[    1.966238] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.966255] usb 1-1.3: Product: RTL2838UHIDIR
[    1.966269] usb 1-1.3: Manufacturer: Realtek
[    1.966284] usb 1-1.3: SerialNumber: 00000001
[    2.066756] usb 1-1.1.3: new full-speed USB device number 5 using dwc_otg
[    2.160639] usb 1-1.1.3: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[    2.160742] usb 1-1.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.160762] usb 1-1.1.3: Product: USB JTAG/serial debug unit
[    2.160778] usb 1-1.1.3: Manufacturer: Espressif
[    2.160794] usb 1-1.1.3: SerialNumber: 8C:BF:EA:DB:5E:D4
[    2.433733] usb 1-1.1.1: new high-speed USB device number 6 using dwc_otg
[    2.522303] usb 1-1.1.1: New USB device found, idVendor=0424, idProduct=7800, bcdDevice= 3.00
[    2.522331] usb 1-1.1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[   13.644045] usbcore: registered new interface driver brcmfmac
[   13.727679] usbcore: registered new interface driver cdc_acm
[   13.804994] usbcore: registered new interface driver lan78xx
```

Guessing the "USB JTAG/serial debug unit" may be the Geosigner.

```
root@fae08cc:~# dmesg | grep -E -c "Under-?voltage detected\!"
0
```

schubydoo, 2025-02-01:
@shawaj Yes I'm all set! Using this exact command on a fresh container reboot allowed me to proceed to flashing with no issues.

`export PIP_EXTRA_INDEX_URL=https://www.piwheels.org/simple && install_packages python3-bitarray python3-cryptography gcc python3-dev`

shawaj, 2025-02-05:
On raspi 4, I get:
```
root@63bf7249a6ae:/tmp# wingbits geosigner flash
Performing full geosigner update...
Downloading firmware...
Creating virtual environment...
Installing esptool...
Collecting esptool
  Downloading esptool-4.8.1.tar.gz (409 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 409.5/409.5 kB 1.5 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting bitstring!=4.2.0,>=3.1.6
  Downloading bitstring-4.3.0-py3-none-any.whl (71 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 71.9/71.9 kB 119.2 kB/s eta 0:00:00
Collecting cryptography>=2.1.4
  Downloading cryptography-44.0.0-cp39-abi3-manylinux_2_34_aarch64.whl (3.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.9/3.9 MB ? eta 0:00:00
Collecting ecdsa>=0.16.0
  Downloading ecdsa-0.19.0-py2.py3-none-any.whl (149 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 149.3/149.3 kB 2.2 MB/s eta 0:00:00
Collecting pyserial>=3.3
  Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 90.6/90.6 kB 1.7 MB/s eta 0:00:00
Collecting reedsolo<1.8,>=1.5.3
  Downloading reedsolo-1.7.0-py3-none-any.whl (32 kB)
Collecting PyYAML>=5.1
  Downloading PyYAML-6.0.2-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (736 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 736.8/736.8 kB 1.7 MB/s eta 0:00:00
Collecting intelhex
  Downloading intelhex-2.3.0-py2.py3-none-any.whl (50 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 50.9/50.9 kB 2.2 MB/s eta 0:00:00
Collecting argcomplete>=3
  Downloading argcomplete-3.5.3-py3-none-any.whl (43 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 43.6/43.6 kB 849.0 kB/s eta 0:00:00
Collecting bitarray<3.1,>=3.0.0
  Downloading bitarray-3.0.0-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (285 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 285.8/285.8 kB 2.1 MB/s eta 0:00:00
Collecting cffi>=1.12
  Downloading cffi-1.17.1-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (469 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 469.2/469.2 kB 1.9 MB/s eta 0:00:00
Collecting six>=1.9.0
  Downloading six-1.17.0-py2.py3-none-any.whl (11 kB)
Collecting pycparser
  Downloading pycparser-2.22-py3-none-any.whl (117 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 117.6/117.6 kB 1.7 MB/s eta 0:00:00
Building wheels for collected packages: esptool
  Building wheel for esptool (pyproject.toml) ... done
  Created wheel for esptool: filename=esptool-4.8.1-py3-none-any.whl size=534238 sha256=21b13b89bc9643bb7dff62f7195035af97a2a4c1342383555a75f1d14cf67478
  Stored in directory: /root/.cache/pip/wheels/f0/39/a7/e213946ae315a441aa182ca5a8a92a3bf5a31f1a60d4559423
Successfully built esptool
Installing collected packages: reedsolo, pyserial, intelhex, bitarray, six, PyYAML, pycparser, bitstring, argcomplete, ecdsa, cffi, cryptography, esptool
Successfully installed PyYAML-6.0.2 argcomplete-3.5.3 bitarray-3.0.0 bitstring-4.3.0 cffi-1.17.1 cryptography-44.0.0 ecdsa-0.19.0 esptool-4.8.1 intelhex-2.3.0 pycparser-2.22 pyserial-3.5 reedsolo-1.7.0 six-1.17.0
Flashing firmware...
esptool.py v4.8.1
Found 2 serial ports
Serial port /dev/ttyAMA0
Connecting...
/dev/ttyAMA0 failed to connect: [Errno 25] Inappropriate ioctl for device
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-C3 in Secure Download Mode
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x003fffff...
Erasing flash...
WARNING: Security features enabled, so not changing any flash settings.
Took 16.03s to erase flash block
Wrote 4194304 bytes at 0x00000000 in 60.9 seconds (550.7 kbit/s)...

Leaving...
Hard resetting via RTS pin...
Firmware flashed successfully
GeoSigner firmware updated successfully
```

Still not sure why @schubydoo you saw different on same device

shawaj, 2025-02-05:
And this was after the manual flash...

```
root@63bf7249a6ae:/tmp# wingbits geosigner info
GeoSigner ID: redacted
GeoSigner Version: geosigner-v1.0.0
Location data: Latitude: redacted, Longitude: redacted, Satellites: 25
✓ GeoSigner is ready to use
root@63bf7249a6ae:/tmp# wingbits feeder start
Started Wingbits client - version: 6b2cf470 device: geosigner built at 2025-02-04 11:12:44 UTC
Wingbits Feeder is listening on port 30006
Found geosigner-v1.0.0 redacted
Updating geosigner...
Progress: 0%  - 0 / 397312 bytes sent
2025/02/05 01:56:29 INFO already at latest version version=6b2cf470
Progress: 18%  - 73728 / 397312 bytes sent
Progress: 36%  - 143360 / 397312 bytes sent
Progress: 61%  - 245760 / 397312 bytes sent
Progress: 85%  - 339968 / 397312 bytes sent
Progress: 100%  - 397312 / 397312 bytes sent
Reconnecting to geosigner...
Geosigner updated successfully
Connected to Wingbits over TLS/TCP
^CBye
root@63bf7249a6ae:/tmp# wingbits geosigner info
GeoSigner ID: redacted
GeoSigner Version: geosigner-1.0.1
Location data: Latitude: redacted, Longitude: redacted, Satellites: 25
✓ GeoSigner is ready to use
root@63bf7249a6ae:/tmp# wingbits feeder start
Started Wingbits client - version: 6b2cf470 device: geosigner built at 2025-02-04 11:12:44 UTC
2025/02/05 02:01:12 INFO next update scheduled for later next_update=2025-02-05T18:00:00Z
Wingbits Feeder is listening on port 30006
Found geosigner-1.0.1 redacted
Connected to Wingbits over TLS/TCP
```

schubydoo, 2025-02-05:
Could it be another permissions type issue because I deployed via Windows?  Can't think of much else that would be different...

shawaj, 2025-02-05:
> Could it be another permissions type issue because I deployed via Windows? Can't think of much else that would be different...

I don't see how it can be that. The things you changed here don't have any effect from permissions https://github.com/ketilmo/balena-ads-b/issues/230#issuecomment-2629048154

There's really nothing I can see that would cause it to have not worked on a Pi4 (or any raspi on 64 bit os for that matter)

shawaj, 2025-02-05:
@jayscovill how's things with yours?

[8 more replies]

closed 2025-02-24 by shawaj