ketilmo/balena-ads-b #246: ADSB-Exchange Can't Find UUID After Recent Update

schubydoo, 2025-01-31:
It's hard to pinpoint when this started as I tend to upgrade and walk away (probably not good practice).  But recently I noticed my adsb-exchange container couldn't located my UUID variable that I set ages ago (and previously worked).  In doing so, the container would just continue to randomly generate a new UUID and not do anything else.

Looking through the README I decided to try adding `ADSB_EXCHANGE_ENABLE_ALL=true` to the variables, and this made adsb-exchange locate my pre-existing UUID and work correctly.

I think this may have been introduced in https://github.com/ketilmo/balena-ads-b/pull/95

Just to clarify, I resolve the issue by using the above variable, but I don't know if others who have upgraded might be stuck in the same situation I was in.  Perhaps there's a startup bug?

Thanks!

shawaj, 2025-01-31:
@schubydoo so you had ADSB_EXCHANGE_SITENAME and ADSB_EXCHANGE_UUID device variables set right, but it still wasn't working or feeding to ADS-B exchange? 

I think I can see the issue but just wanted to confirm the exact issue.

schubydoo, 2025-01-31:
Yes I had both of those variables set since very early on in the balena-ads-b project.  Upon pushing the latest build, the container could no longer locate them.  It only started working again when I used the new variable to enable all of the clones.

shawaj, 2025-01-31:
Thanks @schubydoo - I can see the issue. The logic isn't checking for that correctly since #95 as you said. 

By default it is not starting anything unless you have one of those variables set but it should be using the previous logic of starting just adsb-exchange if you have the variables for it set.

I've got a bit of a migraine coming right now so I'm going to log off but I'll sort this out over the weekend.

Thanks for reporting 🙏

schubydoo, 2025-01-31:
Thanks @shawaj , take care of yourself first :)

shawaj, 2025-02-05:
@schubydoo this should be fixed now. Could you give it a try from your end?

commit cdf621a8 (2025-02-05): adsb-exchange: fix to maintain backward compatibility

In #95 added adsb-exchange clones. However, this broke some existing features of feeding just to ADSB Exchange without needing to add extra device variables as per #246

Closes: #246
PR #255 (2025-02-05): adsb-exchange: fix to maintain backward compatibility
In #95 added adsb-exchange clones. However, this broke some existing features of feeding just to ADSB Exchange without needing to add extra device variables as per #246 

Closes: #246
closed 2025-02-05 by shawaj