ketilmo/balena-ads-b #247: tar1090 in Reboot Loop

schubydoo, 2025-01-31:
Hello!

I recently upgraded to latest and noticed a new container for tar1090.  However it seems that it just continues in a start/stop loop.  I can't keep the terminal opened long enough to look at logs.

```
 tar1090  s6-rc: info: service s6rc-oneshot-runner: starting
 tar1090  s6-rc: info: service s6rc-oneshot-runner successfully started
 tar1090  s6-rc: info: service fix-attrs: starting
 tar1090  s6-rc: info: service libseccomp2: starting
 tar1090  s6-rc: info: service 09-rtlsdr-biastee: starting
 tar1090  s6-rc: info: service fix-attrs successfully started
 tar1090  s6-rc: info: service legacy-cont-init: starting
 tar1090  s6-rc: info: service libseccomp2 successfully started
 tar1090  s6-rc: info: service legacy-cont-init successfully started
 tar1090  [2025-01-31 21:35:47.607][09-rtlsdr-biastee-init] Device type is not rtlsdr, skipping...
 tar1090  s6-rc: info: service startup: starting
 tar1090  s6-rc: info: service 09-rtlsdr-biastee successfully started
 tar1090  [2025-01-31 21:35:47.646][00-startup-checks] s6wrap: bad command: '/etc/s6-overlay/startup.d/00-startup-checks': Permission denied
 tar1090  [2025-01-31 21:35:47.653][startup] Error running /etc/s6-overlay/startup.d/00-startup-checks
 tar1090  s6-rc: warning: unable to start service startup: command exited 1
 tar1090  /run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
 tar1090  /run/s6/basedir/scripts/rc.init: fatal: stopping the container.
 tar1090  s6-rc: info: service libseccomp2: stopping
 tar1090  s6-rc: info: service legacy-cont-init: stopping
 tar1090  s6-rc: info: service 09-rtlsdr-biastee: stopping
 tar1090  s6-rc: info: service libseccomp2 successfully stopped
 tar1090  s6-rc: info: service legacy-cont-init successfully stopped
 tar1090  s6-rc: info: service fix-attrs: stopping
 tar1090  s6-rc: info: service fix-attrs successfully stopped
 tar1090  [2025-01-31 21:35:47.743][09-rtlsdr-biastee-down] Device type is not rtlsdr, skipping...
 tar1090  s6-rc: info: service 09-rtlsdr-biastee successfully stopped
 tar1090  s6-rc: info: service s6rc-oneshot-runner: stopping
 tar1090  s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

I checked the README and it did mention some variables I can set but the defaults seem like they would work, so I'm not sure what specifically I could try.

shawaj, 2025-01-31:
@schubydoo the other containers are working? Just this one stuck like this?

I've not seen this before but now I look at it, if you don't have the adsb-exchange set up it maybe could cause issues. I think that might be the issue here but I need to double check.

If that's the issue then I can fix it very easily just need to confirm

schubydoo, 2025-01-31:
All other containers are working.  I did have some issues with adsb-exchange as mentioned here: https://github.com/ketilmo/balena-ads-b/issues/246

But I fixed those already.

shawaj, 2025-01-31:
And it's on a raspberry pi 4 you said right?

schubydoo, 2025-01-31:
Yep:

Type: Raspberry Pi 4 (using 64bit OS)
OS: balenaOS 6.1.24+rev5
Supervisor: 16.12.0

shawaj, 2025-01-31:
When you deployed the code, did you do it from a windows machine by any chance? Or how did you deploy it?

schubydoo, 2025-01-31:
Ah, yes, I had to manually deploy via balena-cli on Windows using the latest git pull (since the button isn't working as mentioned in the other thread)

```
git clone https://github.com/ketilmo/balena-ads-b
cd .\balena-ads-b\
balena push <<FLEETNAME>>
```

Didn't see any errors.  I could try it again.

newgas, 2025-01-31:
Joining in... exact same issue.

Device type is not rtlsdr... then shutdowns, then restrarts.

Pi3B+ 32bit OS
manually pushed from Windows as per schubydoo

shawaj, 2025-01-31:
It's a permissions issue. The 00-startup-checls is set as an executable file but windows will mess up these permissions. That's totally my fault for forgetting that some people use windows in #190 

Fix incoming

shawaj, 2025-01-31:
Ok @newgas @schubydoo that should be fixed now 🙏

Apologies, I am too used to Mac and Linux permissions and didn't think about manual pushing from Windows 🤦

shawaj, 2025-01-31:
Let me know if it still doesn't work and I'll reopen the issue but I think that should solve it 🤞

closed 2025-01-31 by shawaj

schubydoo, 2025-01-31:
Seems to be starting.  I'm seeing some odd errors, maybe they are benign? I'll need to dig into it a bit more later.

```
tar1090  [2025-01-31 23:40:22.596][graphs1090] 354 (process ID) old priority 0, new priority 19
 tar1090  [2025-01-31 23:40:22.597][readsb] actual range outline, read bytes: 368644
 tar1090  [2025-01-31 23:40:22.599][readsb]  .......... done, loaded 4 aircraft in 0.031 seconds!
 tar1090  [2025-01-31 23:40:22.599][readsb] aircraft table fill: 0.0
 tar1090  s6-svc: fatal: unable to control /run/service/: supervisor not listening
 tar1090  [2025-01-31 23:40:22.604][readsb] 30152: API output port                
 tar1090  [2025-01-31 23:40:22.604][readsb] 
 tar1090  fatal library error, lookup self
 tar1090  [2025-01-31 23:40:22.619][readsb] 30002: Raw TCP output port            
 tar1090  [2025-01-31 23:40:22.628][readsb] 30005: Beast TCP output port          
 tar1090  [2025-01-31 23:40:22.632][readsb] 30006: BeastReduce TCP output port    
 tar1090  [2025-01-31 23:40:22.632][readsb] 30047: Position json output port      
 tar1090  [2025-01-31 23:40:22.633][readsb] 30003: SBS TCP output ALL port        
 tar1090  [2025-01-31 23:40:22.633][readsb] 32006: SBS TCP input MAIN port        
 tar1090  [2025-01-31 23:40:22.634][readsb] 32007: SBS TCP input MLAT port        
 tar1090  [2025-01-31 23:40:22.634][readsb] 32008: SBS TCP input PRIO port        
 tar1090  [2025-01-31 23:40:22.634][readsb] 32009: SBS TCP input JAERO port       
 tar1090  [2025-01-31 23:40:22.635][readsb] 30001: Raw TCP input port             
 tar1090  [2025-01-31 23:40:22.636][readsb] 30004: Beast TCP input port           
 tar1090  [2025-01-31 23:40:22.636][readsb] 30104: Beast TCP input port           
 tar1090  fatal library error, lookup self
 tar1090  s6-svc: fatal: unable to control /run/service/: supervisor not listening
 tar1090  [2025-01-31 23:40:22.660][readsb] Beast TCP input: Connection established: adsb-exchange (172.19.0.12) port 30157
 tar1090  [2025-01-31 23:40:22.660][readsb] Beast TCP input: Connection established: dump1090-fa (172.19.0.6) port 30005
 tar1090  [2025-01-31 23:40:22.709][graphs1090] Generating all graphs
 tar1090  [2025-01-31 23:40:24.032][tar1090-update] tar1090 version: 3.14.1706 tar1090-db version: 3.14.1635
 tar1090  fatal library error, lookup self
 tar1090  fatal library error, lookup self
 tar1090  s6-svc: fatal: unable to control /run/service/: supervisor not listening
 tar1090  [2025-01-31 23:40:27.511][tar1090] 1377 (process ID) old priority 0, new priority 19
 tar1090  [2025-01-31 23:40:35.495][collectd] [2025-01-31 23:40:35] plugin_load: plugin "logfile" successfully loaded.
 tar1090  [2025-01-31 23:40:36.749][collectd] [error] set_thread_name("rrdtool queue"): No such file or directory
 tar1090  [2025-01-31 23:40:36.749][collectd] [error] set_thread_name("writer#0"): No such file or directory
 tar1090  [2025-01-31 23:40:36.749][collectd] [error] set_thread_name("reader#0"): No such file or directory
 tar1090  [2025-01-31 23:40:36.750][collectd] [error] set_thread_name("reader#1"): No such file or directory
 tar1090  [2025-01-31 23:40:36.750][collectd] [error] set_thread_name("reader#2"): No such file or directory
 tar1090  [2025-01-31 23:40:36.750][collectd] [error] set_thread_name("reader#3"): No such file or directory
 tar1090  [2025-01-31 23:40:36.751][collectd] [error] set_thread_name("reader#4"): No such file or directory
```

But /graphs1090/ is working, so probably all ok!

shawaj, 2025-01-31:
I think there's some error with s6 that I couldn't work out that are seemingly benign.

Do you see planes on /tar1090/ endpoint as well? 

If so then I think you are good to go :-)

schubydoo, 2025-02-01:
I do! Thank you.

shawaj, 2025-02-01:
Perfect, I'll move the s6 error error stuff to a new issue and investigate it another time

PR #248 (2025-01-31): fix: tar1090 startup checks permissions when deployed from Windows
When pushing from Windows, file permissions are lost. Meaning 00-startup-checks is no longer executable. This explicitly sets the file permissions. 

Closes #247
commit 7a9e1854 (2025-01-31): fix: tar1090 startup checks permissions when deployed from Windows

When pushing from Windows, file permissions are lost. Meaning 00-startup-checks is no longer executable. This explicitly sets the file permissions. 

Closes #247