ketilmo/balena-ads-b #250: s6 service errors in tar1090 container

shawaj, 2025-02-02:
As per https://github.com/ketilmo/balena-ads-b/issues/247#issuecomment-2628575932

There are several errors coming from the s6 service in the tar1090 container. Unclear why exactly, or whether it's an issue (tar1090 and graphs1090 seem to run ok)

```
tar1090  [2025-01-31 23:40:22.596][graphs1090] 354 (process ID) old priority 0, new priority 19
 tar1090  [2025-01-31 23:40:22.597][readsb] actual range outline, read bytes: 368644
 tar1090  [2025-01-31 23:40:22.599][readsb]  .......... done, loaded 4 aircraft in 0.031 seconds!
 tar1090  [2025-01-31 23:40:22.599][readsb] aircraft table fill: 0.0
 tar1090  s6-svc: fatal: unable to control /run/service/: supervisor not listening
 tar1090  [2025-01-31 23:40:22.604][readsb] 30152: API output port                
 tar1090  [2025-01-31 23:40:22.604][readsb] 
 tar1090  fatal library error, lookup self
 tar1090  [2025-01-31 23:40:22.619][readsb] 30002: Raw TCP output port            
 tar1090  [2025-01-31 23:40:22.628][readsb] 30005: Beast TCP output port          
 tar1090  [2025-01-31 23:40:22.632][readsb] 30006: BeastReduce TCP output port    
 tar1090  [2025-01-31 23:40:22.632][readsb] 30047: Position json output port      
 tar1090  [2025-01-31 23:40:22.633][readsb] 30003: SBS TCP output ALL port        
 tar1090  [2025-01-31 23:40:22.633][readsb] 32006: SBS TCP input MAIN port        
 tar1090  [2025-01-31 23:40:22.634][readsb] 32007: SBS TCP input MLAT port        
 tar1090  [2025-01-31 23:40:22.634][readsb] 32008: SBS TCP input PRIO port        
 tar1090  [2025-01-31 23:40:22.634][readsb] 32009: SBS TCP input JAERO port       
 tar1090  [2025-01-31 23:40:22.635][readsb] 30001: Raw TCP input port             
 tar1090  [2025-01-31 23:40:22.636][readsb] 30004: Beast TCP input port           
 tar1090  [2025-01-31 23:40:22.636][readsb] 30104: Beast TCP input port           
 tar1090  fatal library error, lookup self
 tar1090  s6-svc: fatal: unable to control /run/service/: supervisor not listening
 tar1090  [2025-01-31 23:40:22.660][readsb] Beast TCP input: Connection established: adsb-exchange (172.19.0.12) port 30157
 tar1090  [2025-01-31 23:40:22.660][readsb] Beast TCP input: Connection established: dump1090-fa (172.19.0.6) port 30005
 tar1090  [2025-01-31 23:40:22.709][graphs1090] Generating all graphs
 tar1090  [2025-01-31 23:40:24.032][tar1090-update] tar1090 version: 3.14.1706 tar1090-db version: 3.14.1635
 tar1090  fatal library error, lookup self
 tar1090  fatal library error, lookup self
 tar1090  s6-svc: fatal: unable to control /run/service/: supervisor not listening
 tar1090  [2025-01-31 23:40:27.511][tar1090] 1377 (process ID) old priority 0, new priority 19
 tar1090  [2025-01-31 23:40:35.495][collectd] [2025-01-31 23:40:35] plugin_load: plugin "logfile" successfully loaded.
 tar1090  [2025-01-31 23:40:36.749][collectd] [error] set_thread_name("rrdtool queue"): No such file or directory
 tar1090  [2025-01-31 23:40:36.749][collectd] [error] set_thread_name("writer#0"): No such file or directory
 tar1090  [2025-01-31 23:40:36.749][collectd] [error] set_thread_name("reader#0"): No such file or directory
 tar1090  [2025-01-31 23:40:36.750][collectd] [error] set_thread_name("reader#1"): No such file or directory
 tar1090  [2025-01-31 23:40:36.750][collectd] [error] set_thread_name("reader#2"): No such file or directory
 tar1090  [2025-01-31 23:40:36.750][collectd] [error] set_thread_name("reader#3"): No such file or directory
 tar1090  [2025-01-31 23:40:36.751][collectd] [error] set_thread_name("reader#4"): No such file or directory
```

shawaj, 2025-02-05:
@mikenye do you have any idea what these errors could be caused by? Or if they are an issue?

mikenye, 2025-02-06:
> [@mikenye](https://github.com/mikenye) do you have any idea what these errors could be caused by? Or if they are an issue?

Absolutely no idea :(