ketilmo/balena-ads-b #251: Issues starting containers

D00B1983, 2025-02-02:
Hi all, please excuse the lack of knowledge on this but recently, possibly since the recent Balena updates I am having trouble starting containers.

I have posted the exact errors so apologies for the long first thread.

Each reboot allows different containers to start, so in some cases, FR24 will start and work but Piaware will not etc.

Everything was fine up until recently when I carried out an update on the Host and Supervisor versions.

Any help on this would be much appreciated.

For example - 

**FR24 when trying to start**

Request error: (HTTP code 500) server error - failed to start shim: start failed: runtime/cgo: pthread_create failed: Resource temporarily unavailable SIGABRT: abort PC=0x7fab8749ec m=0 sigcode=18446744073709551610 goroutine 0 [idle]: runtime: unknown pc 0x7fab8749ec stack: frame={sp:0x7feb48cfe0, fp:0x0} stack=[0x7feac8e638,0x7feb48d650) 0x0000007feb48cee0: 0x0000000000402029 0x0000007feb48d108 0x0000007feb48cef0: 0x0000007fabaa36f0 0x000000000f11ed7d 0x0000007feb48cf00: 0x0000007fabaa9000 0x0000007fabaaa360 0x0000007feb48cf10: 0x0000000000000000 0x0000007fabaaa6f8 0x0000007feb48cf20: 0x0000000000000001 0x0000007fab8086d0 0x0000007feb48cf30: 0x0000007f00000005 0x00000000ab819a70 0x0000007feb48cf40: 0x0000000000000006 0x0000007fabaa36f0 0x0000007feb48cf50: 0x00000000003c47b5 0x0000007fabaa3688 0x0000007feb48cf60: 0x0000007fabaa8b10 0x0000007feb48cf88 0x0000007feb48cf70: 0x0000007feb48cf84 0x0000007fab8079c8 0x0000007feb48cf80: 0x00000000eb48cfe0 0x0000000000000000 0x0000007feb48cf90: 0x0000000000000005 0x0000000000000000 0x0000007feb48cfa0: 0x0000007f00000001 0x0000007fabaaa360 0x0000007feb48cfb0: 0x0000007feb48d0b0 0x0000007faba7ffe8 0x0000007feb48cfc0: 0x0000007fabaaa360 0x0000000000000558 0x0000007feb48cfd0: 0x0000000002fd21c8 0x0000000000000001 0x0000007feb48cfe0: <0x0000007feb48d0a0 0x0000007fab83b458 0x0000007feb48cff0: 0x0000000000000006 0x0000007fab96b690 0x0000007feb48d000: 0x0000007faba9e440 0x0000000000000001 0x0000007feb48d010: 0x0000000000000000 0x0000007fabaaa6f8 0x0000007feb48d020: 0x0000000000000000 0x0000007fabaaa360 0x0000007feb48d030: 0x0000000000000000 0x0000007feb48d078 0x0000007feb48d040: 0x0000007fabaa9018 0x0000007fab8086d0 0x0000007feb48d050: 0x0000000000000005 0x00000000abaaa360 0x0000007feb48d060: 0x0000000000000006 0x00000000ffffffff 0x0000007feb48d070: 0x00000000023fe6b0 0x0000007fab814c58 0x0000007feb48d080: 0x0000007fabaa25e0 0x0000007feb48d0a8 0x0000007feb48d090: 0x0000007feb48d0a4 0x0000007fab808080 0x0000007feb48d0a0: 0x0000007feb48d0c0 0x0000007fab82af64 0x0000007feb48d0b0: 0x0000007fab96b000 0x0000007fab82af18 0x0000007feb48d0c0: 0x0000007feb48d210 0x00000000018af4d4 0x0000007feb48d0d0: 0x0000007fab96a5d0 0x0000000001bb28e8 runtime: unknown pc 0x7fab8749ec stack: frame={sp:0x7feb48cfe0, fp:0x0} stack=[0x7feac8e638,0x7feb48d650) 0x0000007feb48cee0: 0x0000000000402029 0x0000007feb48d108 0x0000007feb48cef0: 0x0000007fabaa36f0 0x000000000f11ed7d 0x0000007feb48cf00: 0x0000007fabaa9000 0x0000007fabaaa360 0x0000007feb48cf10: 0x0000000000000000 0x0000007fabaaa6f8 0x0000007feb48cf20: 0x0000000000000001 0x0000007fab8086d0 0x0000007feb48cf30: 0x0000007f00000005 0x00000000ab819a70 0x0000007feb48cf40: 0x0000000000000006 0x0000007fabaa36f0 0x0000007feb48cf50: 0x00000000003c47b5 0x0000007fabaa3688 0x0000007feb48cf60: 0x0000007fabaa8b10 0x0000007feb48cf88 0x0000007feb48cf70: 0x0000007feb48cf84 0x0000007fab8079c8 0x0000007feb48cf80: 0x00000000eb48cfe0 0x0000000000000000 0x0000007feb48cf90: 0x0000000000000005 0x0000000000000000 0x0000007feb48cfa0: 0x0000007f00000001 0x0000007fabaaa360 0x0000007feb48cfb0: 0x0000007feb48d0b0 0x0000007faba7ffe8 0x0000007feb48cfc0: 0x0000007fabaaa360 0x0000000000000558 0x0000007feb48cfd0: 0x0000000002fd21c8 0x0000000000000001 0x0000007feb48cfe0: <0x0000007feb48d0a0 0x0000007fab83b458 0x0000007feb48cff0: 0x0000000000000006 0x0000007fab96b690 0x0000007feb48d000: 0x0000007faba9e440 0x0000000000000001 0x0000007feb48d010: 0x0000000000000000 0x0000007fabaaa6f8 0x0000007feb48d020: 0x0000000000000000 0x0000007fabaaa360 0x0000007feb48d030: 0x0000000000000000 0x0000007feb48d078 0x0000007feb48d040: 0x0000007fabaa9018 0x0000007fab8086d0 0x0000007feb48d050: 0x0000000000000005 0x00000000abaaa360 0x0000007feb48d060: 0x0000000000000006 0x00000000ffffffff 0x0000007feb48d070: 0x00000000023fe6b0 0x0000007fab814c58 0x0000007feb48d080: 0x0000007fabaa25e0 0x0000007feb48d0a8 0x0000007feb48d090: 0x0000007feb48d0a4 0x0000007fab808080 0x0000007feb48d0a0: 0x0000007feb48d0c0 0x0000007fab82af64 0x0000007feb48d0b0: 0x0000007fab96b000 0x0000007fab82af18 0x0000007feb48d0c0: 0x0000007feb48d210 0x00000000018af4d4 0x0000007feb48d0d0: 0x0000007fab96a5d0 0x0000000001bb28e8 goroutine 1 [running, locked to thread]: runtime.systemstack_switch() /usr/lib/go/src/runtime/asm_arm64.s:187 +0x8 fp=0x40000705a0 sp=0x4000070590 pc=0x470588 runtime.newproc(0x0, 0x204ae80) /usr/lib/go/src/runtime/proc.go:4254 +0x54 fp=0x40000705f0 sp=0x40000705a0 pc=0x447c84 runtime.init.7() /usr/lib/go/src/runtime/proc.go:294 +0x30 fp=0x4000070610 sp=0x40000705f0 pc=0x43e530 runtime.doInit(0x30249a0) /usr/lib/go/src/runtime/proc.go:6498 +0x138 fp=0x4000070770 sp=0x4000070610 pc=0x44d228 runtime.main() /usr/lib/go/src/runtime/proc.go:204 +0x13c fp=0x40000707d0 sp=0x4000070770 pc=0x43e1fc runtime.goexit() /usr/lib/go/src/runtime/asm_arm64.s:1133 +0x4 fp=0x40000707d0 sp=0x40000707d0 pc=0x4729d4 r0 0x0 r1 0x25a5 r2 0x6 r3 0x8 r4 0x20 r5 0x7feb48ac77 r6 0x80ffce12630000 r7 0xffffffff r8 0x83 r9 0x7fab920280 r10 0x7fab8079c8 r11 0x7feb48cf88 r12 0x7feb48cf84 r13 0x0 r14 0x21cc30a0 r15 0x0 r16 0x0 r17 0x0 r18 0x270f r19 0x25a5 r20 0x6 r21 0x7faba9e440 r22 0x1 r23 0x0 r24 0x2b186d0 r25 0x4041b4 r26 0x7feb48d430 r27 0x0 r28 0x30abd00 r29 0x7feb48cfe0 lr 0x7fab8749d8 sp 0x7feb48cfe0 pc 0x7fab8749ec fault 0x0 : exit status 2: unknown

**PiAware - if fails to start after reboot**

Request error: (HTTP code 500) server error - failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error running hook #0: error running hook: exit status 2, stdout: , stderr: runtime/cgo: pthread_create failed: Resource temporarily unavailable SIGABRT: abort PC=0x7faa9f49ec m=0 sigcode=18446744073709551610 goroutine 0 [idle]: runtime: unknown pc 0x7faa9f49ec stack: frame={sp:0x7fe32bee20, fp:0x0} stack=[0x7fe2ac03f8,0x7fe32bf410) 0x0000007fe32bed20: 0x0000000000402029 0x0000007fe32bef48 0x0000007fe32bed30: 0x0000007faac276f0 0x000000000f11ed7d 0x0000007fe32bed40: 0x0000007faac2d000 0x0000007faac2e360 0x0000007fe32bed50: 0x0000000000000000 0x0000007faac2e6f8 0x0000007fe32bed60: 0x0000000000000001 0x0000007faa9886d0 0x0000007fe32bed70: 0x0000007f00000001 0x00000000aa999a70 0x0000007fe32bed80: 0x0000000000000006 0x0000007faac276f0 0x0000007fe32bed90: 0x00000000003c47b5 0x0000007faac27688 0x0000007fe32beda0: 0x0000007faac2cb10 0x0000007fe32bedc8 0x0000007fe32bedb0: 0x0000007fe32bedc4 0x0000007faa9879c8 0x0000007fe32bedc0: 0x00000000e32bee20 0x0000000000000000 0x0000007fe32bedd0: 0x0000000000000001 0x0000000000000000 0x0000007fe32bede0: 0x0000007f00000001 0x0000007faac2e360 0x0000007fe32bedf0: 0x0000007fe32beef0 0x0000007faac03fe8 0x0000007fe32bee00: 0x0000007faac2e360 0x0000000000000558 0x0000007fe32bee10: 0x0000000002fd21c8 0x0000000000000001 0x0000007fe32bee20: <0x0000007fe32beee0 0x0000007faa9bb458 0x0000007fe32bee30: 0x0000000000000006 0x0000007faaaeb690 0x0000007fe32bee40: 0x0000007faac22440 0x0000000000000001 0x0000007fe32bee50: 0x0000000001bc3e98 0x0000007faac2e6f8 0x0000007fe32bee60: 0x0000000000000000 0x0000007faac2e360 0x0000007fe32bee70: 0x0000000000000000 0x0000007fe32beeb8 0x0000007fe32bee80: 0x0000007faac2d018 0x0000007faa9886d0 0x0000007fe32bee90: 0x0000000000000001 0x00000000aac2e360 0x0000007fe32beea0: 0x0000000000000006 0x00000000ffffffff 0x0000007fe32beeb0: 0x00000000023fe6b0 0x0000007faa994c58 0x0000007fe32beec0: 0x0000007faac265e0 0x0000007fe32beee8 0x0000007fe32beed0: 0x0000007fe32beee4 0x0000007faa988080 0x0000007fe32beee0: 0x0000007fe32bef00 0x0000007faa9aaf64 0x0000007fe32beef0: 0x0000007faaaeb000 0x0000007faa9aaf18 0x0000007fe32bef00: 0x0000007fe32bf050 0x00000000018af4d4 0x0000007fe32bef10: 0x0000007faaaea5d0 0x0000000001bb28e8 runtime: unknown pc 0x7faa9f49ec stack: frame={sp:0x7fe32bee20, fp:0x0} stack=[0x7fe2ac03f8,0x7fe32bf410) 0x0000007fe32bed20: 0x0000000000402029 0x0000007fe32bef48 0x0000007fe32bed30: 0x0000007faac276f0 0x000000000f11ed7d 0x0000007fe [... cut ...]
 piaware  faup1090 exited with EXIT 1
 piaware  will reconnect to the ADS-B data program at dump1090-fa/30005 in 30 seconds

**Radarbox tries to start...**

radarbox  Receiver host is set: dump1090-fa
 radarbox  Receiver port is set: 30005
 radarbox   
 radarbox  Settings verified, proceeding with startup.
 radarbox   
 radarbox  System Architecture: arm64
 radarbox  getaddrinfo: Temporary failure in name resolution
 radarbox  [2025-02-02 18:25:24]  Could not resolve host name specified in 'external_host'.

D00B1983, 2025-02-02:
planefinder  2025-02-02 23:24:22.185494 [V] Failed to resolve remote socket address dump1090-fa 30005: No address associated with hostname (errno: -5)
 planefinder  2025-02-02 23:24:22.188920 [-] Failed TCP connection to all provided addresses for: dump1090-fa:30005

shawaj, 2025-02-03:
It seems to me that maybe you are running on an unsupported architecture. 

What device are you trying to run this on?

D00B1983, 2025-02-03:
Hi, this is a Raspberry Pi 4. Same one I've used for over 12 months.

shawaj, 2025-02-04:
@D00B1983 ah ok. How did you deploy the update to it?

D00B1983, 2025-02-04:
> [@D00B1983](https://github.com/D00B1983) ah ok. How did you deploy the update to it?

The usual way for me, checking these are up to date and if not, click the pencil icon next to them to update

[image]

D00B1983, 2025-02-04:
This is all I am seeing now on the console 

[image]

shawaj, 2025-02-04:
How did you push the updated containers to the device? I mean the application, not the host OS and supervisor.

D00B1983, 2025-02-04:
I have not done that yet, this is a the build that is installed from the guide on this page

shawaj, 2025-02-04:
I would open a terminal to the host OS from the device page, and then enter the following commands: 

```
systemctl stop balena-supervisor
systemctl stop balena
rm -rf /var/lib/balena/{aufs,overlay,containers,image,tmp}
systemctl start balena
update-balena-supervisor
```

This will delete all containers and redownload them.

shawaj, 2025-02-04:
> I have not done that yet, this is a the build that is installed from the guide on this page

I think you should probably rebuild it for the new OS. You may have some success with the above process. 

But if not then I would push an up to date version of the application containers.

Lastly, given the word error messages, there's a chance youe SD card could be corrupted or dying. How old is the SD card?

D00B1983, 2025-02-04:
[image]

It has begun, I'll report back. The SD card is about 12 months old. I honestly didnt give that a thought and something I could change and try tomorrow

D00B1983, 2025-02-04:
[image]

So far so good...

shawaj, 2025-02-04:
Hopefully that might fix it. Otherwise maybe try re-deploying the app.

12 months on the SD card should be fine. Usually they'll be ok for at least a few years.

D00B1983, 2025-02-04:
[image]

Dump1090-fe is having trouble starting by the looks of things

shawaj, 2025-02-04:
Has the SDR dongle come loose perhaps?

And have you plugged any new USB devices in recently or changed power supply?

D00B1983, 2025-02-04:
> Has the SDR dongle come loose perhaps?
> 
> And have you plugged any new USB devices in recently or changed power supply?

Not that I can tell. 
Nothing new added either.

Something I have been thinking, would it be possible to run this on the Raspberry from a USB hard drive and not a SD Card as per this instructions? 
I've been looking at the possibility but not for this instance.

shawaj, 2025-05-22:
@D00B1983 did you get this solved in the end?

D00B1983, 2025-05-31:
Hi @shawaj Yes, the solution was to bin the sd card and install on a new one.