ketilmo/balena-ads-b #296: wingbits problem

RudiZhang, 2025-03-24:
wingbits  2025-03-24T00:30:11.304707Z  WARN sink{component_kind="sink" component_id=vector component_type=vector}:request{request_id=1}: vector::sinks::util::retries: Request timed out. If this happens often while the events are actually reaching their destination, try decreasing `batch.max_bytes` and/or using `compression` if applicable. Alternatively `request.timeout_secs` can be increased. internal_log_rate_limit=true

shawaj, 2025-03-24:
@RudiZhang you need to update to latest version. Wingbits does not use vector anymore since a few months. 

You can follow the instructions here - https://github.com/ketilmo/balena-ads-b?tab=readme-ov-file#part-16--updating-to-the-latest-version

RudiZhang, 2025-04-01:
> [@RudiZhang](https://github.com/RudiZhang) you need to update to latest version. Wingbits does not use vector anymore since a few months.
> 
> You can follow the instructions here - https://github.com/ketilmo/balena-ads-b?tab=readme-ov-file#part-16--updating-to-the-latest-version

ah indeed

closed 2025-04-01 by RudiZhang