ketilmo/balena-ads-b #297: Unable to run showkey.sh script in airnav-radar

Haldaug, 2025-03-24:
While setting up `airnav-radar` for the first time, the `showkey.sh` script did not run. I got the following error message: 

[image]

A workaround is to run rbfeeder directly using this code: `rbfeeder --showkey --no-start`

shawaj, 2025-03-24:
the command is just `/showkey.sh` (it is in the root folder) - you do not need the .. in front

shawaj, 2025-03-24:
@Haldaug see above

Haldaug, 2025-03-24:
The `rm /tmp/rbfeeder.log` line in `showkey.sh` still fails when running it for the first time because there is no `rbfeeder.log` file to remove.

shawaj, 2025-03-24:
If your just run /showkey.sh with the slash in front it doesn't matter what your present working directory is.

I guess we can add -f to the script so it runs even if no log.

Can you edit showkey.sh so the `rm /tmp/rbfeeder.log` on line 14 changes to `rm -f /tmp/rbfeeder.log`

Then it should work I think.

commit 1421fca9 (2025-03-26): airnav: fix showkey script

Fix showkey script

Closes: #297
closed 2025-03-24 by shawaj