ketilmo/balena-ads-b #307: AirNav-Radar Issue

Care-Less, 2025-03-30:
The container is stuck on a loop... Running on a Raspberry Pi 5

**Fault:** airnav-radar  /start.sh: line 98:    17 Segmentation fault      /usr/bin/rbfeeder

**Complete boot log:**

2025-03-30T01:22:10+00:00  airnav-radar  AirNav Radar key is set: ---
2025-03-30T01:22:10+00:00  airnav-radar  Receiver latitude is set: ---
2025-03-30T01:22:10+00:00  airnav-radar  Receiver longitude is set: ---
2025-03-30T01:22:10+00:00  airnav-radar  Receiver altitude is set: ---
2025-03-30T01:22:10+00:00  airnav-radar  Receiver host is set: dump1090-fa
2025-03-30T01:22:10+00:00  airnav-radar  Receiver port is set: 30005
2025-03-30T01:22:10+00:00  airnav-radar   
2025-03-30T01:22:10+00:00  airnav-radar  Settings verified, proceeding with startup.
2025-03-30T01:22:10+00:00  airnav-radar   
2025-03-30T01:22:10+00:00  airnav-radar  System Architecture: arm64
2025-03-30T01:22:10+00:00  airnav-radar  /start.sh: line 98:    17 Segmentation fault      /usr/bin/rbfeeder
2025-03-30T01:22:11+00:00 Service exited 'airnav-radar sha256:1becabc7ab49d3a95da28be3748c1e27c2bcd99f078bc7e69fac9526e7c7f068'

ketilmo, 2025-03-30:
Hi @Care-Less,

Thanks for reaching out, and I'm sorry for the inconvenience. This problem was identified a few days ago and has now been fixed through #306. Simply redeploy the repo to your device, and it should work again. Let us know how it goes!

Care-Less, 2025-03-31:
Thanks @ketilmo it seems to be working without faults after redeploying!

ketilmo, 2025-03-31:
Thanks for reporting back, @Care-Less! Much appreciated. I'm happy to hear that it works now. Happy feeding!